Telekatz/venus.dbus-hoymiles #1: Mehrere Inverter

VOPAtec, 2022-11-04:
Hallo Telekatz,
das ist wirklich ein super Ansatz, die Inverter direkt in Venus zu integieren. Meine Frage ist, wie das ganze mit mehreren , z.B. 3 Stück HM1500 an einer DTU, aussehen würde. Würde dies schon funktionieren?
Vielen Dank und Grüße
Volker

Telekatz, 2022-11-04:
Nein, mehr als ein Inverter wird aktuell nicht unterstützt.

VOPAtec, 2022-11-14:
Schade, besteht denn die Chance, dass es irgendwann kommt? 
Ich finde es bis jetzt die schönste Lösung, da die Konfiguration über die Console vom Cerbo gemacht wird.
Ich versuche es gerade mit [dbus-mqtt-devices] allerdings scheitere ich an der Configuration von Tasmota nach dem Registration Protocol.
Aktuell habe ich nur einen HM1200 laufen. Würde das "out of the box" funktionieren?
Vielen Dank und Grüße

Telekatz, 2022-11-14:
Wird vielleicht noch kommen. Hab selber aber auch nicht mehrere Inverter, um das testen zu können.
Ein HM1200 sollte funktionieren.

closed 2023-02-14 by Telekatz