Telekatz/venus.dbus-hoymiles #3: After installing, all values in VenusOS are 0 - including battery voltage, causing alarm

mcleinn, 2023-02-22:
Hello, I installed the code as described, after which most values in Venus are zeroed - causing a battery alarm. So I quickly uninstalled the code, and everything went back to normal. Do you have any idea why this could happen, and how to install the code safely? Thank you

Telekatz, 2023-02-23:
What version of Venus OS are you using and what devices does your installation include?

mcleinn, 2023-02-23:
Firmware is v.2.92

Quattro II 48/5000
Pylontech batteries
3 Victron MPPTs
Fronius 5.0-1 on AC-OUT

To be connected
OpenDTU (current version)
Hoymiles HM-800

Thank you

Telekatz, 2023-02-23:
It doesn't work because you already have a Quattro II in your installation.

This script is for a Hoymiles inverter connected to a battery. It emulates a Quattro inverter to be able to activate the ESS assistant. The system gets confused with the two Quattro devices.

I think this script is more suitable for your system:
[https://github.com/henne49/dbus-opendtu](https://github.com/henne49/dbus-opendtu)

mcleinn, 2023-02-23:
Thank you, indeed, I did not realise that your script is intended for a
different type of system.

I thought "battery connected" meant the DC-side of the Hoymiles is
connected to a solar panel, and the AC-side to the battery* via a MultiPlus
II or Quattro II. *

Now I understand the DC-side of your HM-600 is connected *directly *to the
battery *instead *of a solar-panel, basically like a MPII without the
charger.

That is a pretty interesting setup. Is there any website describing the
hardware side of the project?

Maybe it would be worth adding a sentence at the top to avoid
misunderstanding.

Best regards

Telekatz, 2023-02-24:
There is no website describing my system, but the components used in my installation are listed in the readme file.

closed 2023-03-03 by Telekatz