Telekatz/venus.dbus-hoymiles #4: (PR) Allow external control of HM

ude6, 2023-03-10:
Dear Telekatz,
thank you for this great project. I am using a similar setup as yours but use Node-Red for the control 
of the HM via Open-DTU. Would it be possible to implement another "limit Mode" that would disable the 
steering of the HM via your driver and only leave the display of the values in the cerbo?

Thanks
Andreas

Telekatz, 2023-03-10:
Would a limit mode help you, where you can specify a limit externally via the dbus?

ude6, 2023-03-10:
I guess that would also do the trick. I could then post to the dbus from Node-Red instead 
of going directly to MQTT-->Open-DTU...

TheGrumpyBadger, 2023-03-11:
Yes, that would be very handy. Thx.

Telekatz, 2023-03-11:
I created a new branch to test this feature.
After selecting the "External" feed-in limit mode, the limit can now be set by writing the value to the path "/Ac/PowerLimit" of the dbus service "com.victronenergy.hm".

ude6, 2023-03-11:
That was quick :-) 
I have to set-up a Raspi to test this. Can not take "production" down and risk the running system.
I will need a few days :-(

TheGrumpyBadger, 2023-03-11:
Is this "external" branch based on VenusOS v2.9 or already for v3.00 ?

Telekatz, 2023-03-11:
The External_control branch is for V2.92.

closed 2023-10-22 by Telekatz