Telekatz/venus.dbus-hoymiles #7: ESS Settings

Head, 2023-04-27:
Hi, ich habe heute mit den Victron settings>ESS Einstellungen rumgespielt und war verwundert, warum mein Hoymiles auf Maximum läuft, obwohl ich "Maximum inverter power" und "Grid setpoint" eingestellt habe.
Hat eine Weile gedauert, bis ich gemerkt habe, dass die Einstellungen dort gar nichts machen?
Auf jeden Fall hat mein Hoymiles sein Limit nicht mehr geändert, bis ich dann im ESS den Mode weg von "Optimized (with BatteryLife)" gestellt hab. Erst als ich dann hier "Optimized (without BatteryLife)" gesagt habe, hat dein Script den Hoymiles wieder auf 30W geregelt. (Warum eigentlich 30W, geht nicht weniger?)

Sprich: Welche Einstellungen unter "ESS" nimmt dein Script und welche werden ignoriert? Ich dachte der Vorteil von dem Venus OS (gegenüber openDTUonBattery) sei, dass da "jemand" (also Victron) mit KnowHow und Sicht auf das BMS und die Batterie-Lebensdauer den Laderegler und den Inverter steuert. Quasi das Venus OS denkt, dass der Hoymiles ein "normaler" Victron-Wechselrichter sei und diesen fernsteuert.

(Nebenbei verstehe ich die Einstellungen unter "Startup" nicht und die "Tolerance" ist mir auch nicht so ganz klar.)

Telekatz, 2023-04-27:
Venus OS kümmert sich ja auch um die Kommunikation zwischen BMS und Laderegler und steuert den Laderegler entsprechend den Vorgaben vom BMS. Auch wird von Venus OS anhand vom SOC gesteuert, ob der Inverter produzieren darf oder nicht. Dies wird im ESS Menü über die Einstellungen für "Modus" und "SOC Mindestwert Entladung" eingestellt.

Im ESS Menü wird momentan außerdem die Einstellung für "Wechselrichterleistung begrenzen" von meinem Script berücksichtigt. Damit kannst du die Maximalleistung des Inverters auf 800W drosseln. Was ich noch vorhabe ist die Einstellung für "DC-gekoppelte PV-Überschusseinspeisung" zu berücksichtigen und den "Sollwert Netz" anstelle der Einstellung "Grid Target Power" in den Einstellungen vom Script zu verwenden.

Auf 30W wird der Inverter beim Starten limitiert. Sobald das Script erkennt, dass der Inverter auch läuft, sollte dann anhand des eingestellten "Feed-In Limit Mode" die Leistung geregelt werden.

Die Einstellung für "Startup" regelt die Maximalleistung des Inverters nachdem er eingeschalten wurde anhand der momentanen Solarleistung. Ich habe dies Funktion für die Zeiten eingebaut, in denen nicht viel Leistung von den Solarzellen kommt. Würde der Inverter mehr Leistung aufnehmen, als im Durchschnitt Solarleistung vorhanden ist, würde er immer wieder ein- und ausschalten, da die Batterie immer wieder leer gesaugt wird.

Die Einstellungen für Tolerance geben ausgehend von eingestellten Leistung "Grid Target Power" die untere und obere Abweichung vor, die vom Netz bezogen werden darf, bevor wieder nachgeregelt wird.

Head, 2023-05-05:
Ok, zusammen mit https://github.com/Telekatz/venus.dbus-hoymiles/issues/8 habe ich es jetzt (denke ich) verstanden. Auf einmal macht das auch total Sinn, was in der Readme steht. Bis auf das `Startup Limit Min`.
(Die Anlage läuft auch immer noch nicht in Betrieb, nur Trockenlauf am Akku ohne AC/DC. Aber evt perfekt fürs Debuggen.)

Ich glaube ich habe da einen Bug entdeckt:
In den ESS Settings habe ich `Optimized (with BatteryLife)`. Batterie ist neu und auf 50%. Hier im Script ist alles an, Startup, Feed-In Grid Target. Jetzt schalte ich die Batterie an, Hoymiles fährt hoch und springt auf sein "dauerhaftes Limit" (Also initial sein Max) und ignoriert alle Maxima, die ich in ESS oder deinem Script eingestellt habe. Dein Script regelt jetzt auch nichts nach. Stelle ich auf `Optimized (without BatteryLife)` regelt er. Zurück auf `Optimized (with BatteryLife)` keine Regelung mehr und das Limit bleibt stehen, wo es zu dem Zeitpunkt stand.
Oder möchte das Venus OS den neu gekauften Akku für's BatteryLife erstmal gerne komplett entladen?

2. Bug?
ESS zurück auf `Optimized (without BatteryLife)`, dass er wieder regelt. Jetzt schalte ich den Akku aus, der Wechselrichter geht aus. 2 Minuten warten. Akku an, Wechselrichter fährt hoch, ich beobachte openDTU, jetzt sollte ja das Startup Limit greifen, weil PV ist ja 0 bei mir und der Wechselrichter sollte auf 0, weil 0 PV gehen? Nope. Das Script regelt aufs Grid. (Ist ja theoretisch auch richtig, weil ich jetzt nachts die Anlage neu gestartet hatte und den Akku bis morgen ja leer haben will, aber dann stimmt das "Startup" irgendwie doch wieder nicht?
Oder ist hier das `Startup Limit Min` entscheidend?

Telekatz, 2023-05-05:
Wenn der Inverter Spannung bekommt startet er von sich aus und, sofern kein persistentes Limit eingestellt wurde, mit voller Leistung. In dem Fall ist das Script wohl durcheinander gekommen, weil es annahm, dass der Inverter aus ist. Sollte eigentlich erkannt werden. Muss ich nachschauen. Eventuell musst du auch nur länger warten, bis das erkannt und korrigiert wird.
Wie hoch war der Wert `Tatsächl. Status d. Ladebegrenz.` mit BatteryLife?
Venus OS selber gibt nur vor, ob der Inverter aktiv sein darf. Das Limit kommt allein vom Script.

Zu 2.
Das ist so beabsichtigt. Startup limit ist nur aktiv, wenn beim starten des Inverters auch PV Energie vorhanden ist. Ansonsten wird es gleich deaktiviert. Eben für den Fall, dass der Inverter Nachts gestartet wird.
`Startup Limit Min` gibt an, welche Leistung mindestens bei aktivierten Startup Limit möglich ist.

Head, 2023-06-01:
Ich mach das hier mal zu, meine Fragen sind beantwortet und meine Anlage läuft problemlos. `With BatteryLife` soll man wohl auch nicht verwenden, weil es mit LiFePO4 nur Probleme macht. Mittlerweile gab es von dir ja auch ein Update mit der `Venus OS`-Regelung, die speist auch schön den Überschuss ein, wenn der Akku voll ist. Kann mich noch nicht so ganz entscheiden ob `Base Load` oder `Venus OS` besser ist. Ich glaube fast beim Akku entladen, wenn man den eh leer bekommt, ist `Base Load` besser. Aber zum Überschuss einspeisen und so wenig wie möglich aus dem Netz beziehen `Venus OS`. (Unterschiede von `Venus OS` zu `Grid Target` habe ich bisher kaum/keine bemerkt.)

closed 2023-06-01 by Head

Telekatz, 2023-06-01:
`Venus OS` und `Grid Target` machen im Grunde das selbe, nämlich die vom Netz Bezogene Leistung auf den `Sollwert Netz` zu regeln. 

Bei `Venus OS` kommt die Regelung von Venus OS. Hier wird ständig nachgeregelt, teilweise im zwei Sekundentakt. Bei `Grid Target` ist die Regelung im Script implementiert. Da wird nur nachgeregelt, wenn der Netzwert einen einstellbaren Bereich verlässt.

Head, 2023-06-07:
Habe übrigens festgestellt, dass es fiese hin und her Regelungen (+50, -70, +90, -110, ...) gibt, wenn man das Grid sekündlich übermittelt, aber der HM 3 Sekunden hin und 5 zurück braucht. Wenn man das Grid alle 10 Sekunden schickt ist alles in Ordnung, eine 100% saubere Nulleinspeisung darf man mit openDTU ja eh nicht wirklich erwarten.