Telekatz/venus.dbus-hoymiles #8: support with the settings

recepucar, 2023-05-02:
I'm new to VictronOS and Solar.
My setup: Two PV modules, Victron SmartSolar MTTP 150/35, battery with 3.5 KWH, Raspi with VictronOS, Hoymiles HM-600, Shelly 3em for the load indicator.
So far everything is fine, everything is running.
First I tried OpenDtu later I tried it with OpenDTU on battery, unfortunately not as I wanted.
Today I came across your script, but I didn't really understand the settings and I couldn't find any instructions either.
Your script takes data from the ESS and which one adjusts.
I don't know if it's possible.
I would first charge the battery until its SOC is 90%, later when the battery is full, I think it can feed in a full 600w or what comes from the charge controller, but if the charge controller no longer brings anything, i.e. no more sun, it should be with 150w keep running until the battery runs out.

I would be very happy if you would help me, I think that these settings would help many people.

Telekatz, 2023-05-02:
My script does not provide a regulation as intended by you, in which the battery should first be charged to 90%. It does not decide whether the inverter is allowed to run or not. This is specified by Venus OS based on the ESS settings. The relevant settings in the ESS menu are "Mode" (Optimized with or without BatteryLife) and "Minimum Discharge SoC". Venus OS stops the inverter when the SOC reaches the minimum SOC and restarts the inverter at minimum SOC +3.

Full charge of the battery before running the inverter would also reduce the efficiency of the system, since storing the energy in the battery means additional losses. It is more efficient to feed in the energy that is currently required immediately and only store the excess energy in the battery. That's what my script does.

recepucar, 2023-05-02:
As I said, I'm a solar newbie, it's logical to feed in when the sun is there.
Can you set it so that I can take the network reference from the shelly 3em. When the sun is there, my basic needs are covered and I charge the rest of the battery at the same time

recepucar, 2023-05-02:
sorry, i'm a bit confused.

Maximum inverter power: 600w
Startup Limit: ON
Startup Limit Min: 100w
Startup Limit Max: 600w ??
Feed-In Limit Mode: Base Load
Base Load Period: 5 mins
Base load minimum interval: 5 seconds

Victron ESS settings:
Mode: Optimized (Without Battery Life)
Inverter AC output in use: ON
Minimum SOC:10%
Limit charger power: OFF
Limit inverter power: OFF
Grid setpoint:50w
Grid feed in:
AC - coupled PV: ON
DC - coupled PV: ON

That's how it looks to me. I didn't quite understand Startup Limit Max.
Should I still improve my attitude?

recepucar, 2023-05-06:
a big thank you for your project. It's going great.
I only have one small problem.
Before your project, I used the dbus-opendtu from henne49, since the Victron VRM displayed history data >> consumption (displayed from grid, from solar, consumption) or solar (displayed TO battery, direct use, solar).
and now From Solar and Direct use are no longer displayed.
I don't know if I can set this somewhere or if it doesn't work.

Telekatz, 2023-05-07:
It doesn't work for me in VRM either, but I don't know why.

recepucar, 2023-05-08:
I don't know but I noticed something this morning. Settings > System Setup > Battery Measurements (Shown as Battery on my HM-600.
and VRM, the HM-600 is also marked with a battery icon.

Telekatz, 2023-05-08:
This command is used to get a unique value for DeviceInstance. The script doesn't use this command because it uses a fixed value from the config.ini file.

recepucar, 2023-05-08:
Codes aren't my thing, I'm more of a maker.
in any case, everything runs reliably without bugs.
I found a bug, namely the daily counter in Hoymiles is not reset. maybe you could program it in such a way that it makes a restet the hoymiles at midnight

closed 2023-06-01 by recepucar