Telekatz/venus.dbus-hoymiles #9: Script läuft nach Stromausfall nicht wieder an

TomBo1969, 2023-06-03:
Moin!
Zuerst einmal vielen Dank für das tolle Script, es trifft genau meinen Anwendungsfall und ist super!

Folgendes Problem:
Nach einem Stromausfall/dem Ziehen der Sicherung läuft das Script nicht wieder an, womit letztlich die Akkus leer gezogen werden, bis das BMS eingreift.
Auch wenn ich die Komponenten nach Einsatz der Sicherung von Hand in vorgesehener Reihenfolge wieder einschalte, bleibt im Venus OS (v.2.94) folgendes Bild (dort sind auch meine Komponenten erkennbar (HM-300 ist ein kleines Panel, welches nicht über Dein Script gesteuert wird)) :
[image]

Auch nach Reboot des Venus OS ändert sich nichts, erst wenn ich deinen Service per SSH neu starte/neu installiere, läuft er wieder.
Beim Versuch des unistall.sh kommt noch folgender Fehler:
[image]

Lässt sich das Ganze besser abfangen?
Danke und Gruß,
Tom

Telekatz, 2023-06-03:
Was steht nach einem Reboot in der Datei /data/rc.local?

TomBo1969, 2023-06-03:
/data/rc.local:

#!/bin/bash`

/data/dbus-shelly-3em-smartmeter/install.sh
/data/dbus-opendtu/install.sh
/data/dbus-opendtu2/install.sh
/data/dbus-shelly-1pm-pvinverter/install.sh
/data/dbus-hoymiles/install.sh

Aus dem current.log sollte das der relevante Zeitraum sein:

2023-06-03` 11:25:54,610 root INFO dbus_value_changed: com.victronenergy.system /Dc/Battery/Soc 53
2023-06-03 11:27:56,536 dbusmonitor INFO com.victronenergy.system disappeared from the dbus. Removing it from our lists
2023-06-03 11:27:56,709 root INFO dbus device removed: com.victronenergy.system 0
2023-06-03 11:27:56,793 dbusmonitor INFO com.victronenergy.hub4 disappeared from the dbus. Removing it from our lists
2023-06-03 11:27:56,794 root INFO dbus device removed: com.victronenergy.hub4 0
2023-06-03 11:27:56,795 dbusmonitor INFO com.victronenergy.settings disappeared from the dbus. Removing it from our lists
2023-06-03 11:27:56,796 root INFO dbus device removed: com.victronenergy.settings 0
2023-06-03 11:28:26,573 root INFO Start
2023-06-03 11:28:26,580 root INFO Connected to dbus, and switching over to gobject.MainLoop() (= event based)
2023-06-03 11:28:26,620 dbusmonitor INFO ===== Search on dbus for services that we will monitor starting... =====
2023-06-03 11:28:26,624 dbusmonitor INFO Found: com.victronenergy.settings, scanning and storing items
2023-06-03 11:28:26,626 dbusmonitor INFO        com.victronenergy.settings has device instance 0
2023-06-03 11:28:26,711 dbusmonitor INFO Found: com.victronenergy.system, scanning and storing items
2023-06-03 11:28:27,15 dbusmonitor INFO        com.victronenergy.system has device instance 0
2023-06-03 11:28:27,59 dbusmonitor INFO ===== Search on dbus for services that we will monitor finished =====
2023-06-03 11:28:27,201 root INFO Setting /Settings/SystemSetup/AcInput1 does not exist yet or must be adjusted
2023-06-03 11:28:27,275 root INFO Setting /Settings/SystemSetup/AcInput2 does not exist yet or must be adjusted
2023-06-03 11:28:27,361 root INFO registered ourselves on D-Bus as com.victronenergy.hm
2023-06-03 11:28:27,509 root ERROR Fehler beim connecten mit Broker
Traceback (most recent call last):
  File "/data/dbus-hoymiles/HMpvinverter.py", line 555, in _init_MQTT
    self._MQTTclient.connect(self.settings['/MqttUrl'])  # connect to broker
  File "/usr/lib/python3.8/site-packages/paho/mqtt/client.py", line 941, in connect
    return self.reconnect()
  File "/usr/lib/python3.8/site-packages/paho/mqtt/client.py", line 1075, in reconnect
    sock = self._create_socket_connection()
  File "/usr/lib/python3.8/site-packages/paho/mqtt/client.py", line 3546, in _create_socket_connection
    return socket.create_connection(addr, source_address=source, timeout=self._keepalive)
  File "/usr/lib/python3.8/socket.py", line 808, in create_connection
    raise err
  File "/usr/lib/python3.8/socket.py", line 796, in create_connection
    sock.connect(sa)
OSError: [Errno 101] Network is unreachable
2023-06-03 11:28:27,574 root INFO registered ourselves on D-Bus as com.victronenergy.vebus.DTU_id51
2023-06-03 11:28:27,787 root INFO setting changed, setting: /Settings/SystemSetup/AcInput1, old: 1, new: 1
2023-06-03 11:28:27,815 root INFO setting changed, setting: /Settings/SystemSetup/AcInput2, old: 0, new: 0
2023-06-03 11:28:27,963 dbusmonitor INFO Found: com.victronenergy.hub4, scanning and storing items
2023-06-03 11:28:27,988 dbusmonitor INFO        com.victronenergy.hub4 has device instance 0
2023-06-03 11:28:27,995 root INFO dbus device added: com.victronenergy.hub4 0
2023-06-03 11:29:06,682 dbusmonitor INFO Found: com.victronenergy.acload.http_47, scanning and storing items
2023-06-03 11:29:06,711 dbusmonitor INFO        com.victronenergy.acload.http_47 has device instance 47
2023-06-03 11:29:06,758 root INFO dbus device added: com.victronenergy.acload.http_47 47
2023-06-03 11:29:08,124 root INFO dbus_value_changed: com.victronenergy.hub4 /PvPowerLimiterActive 1
2023-06-03 11:30:38,629 root INFO dbus_value_changed: com.victronenergy.system /Dc/Battery/Soc 54
2023-06-03 11:35:05,73 root INFO dbus_value_changed: com.victronenergy.system /Dc/Battery/Soc 55
2023-06-03 11:39:40,442 root INFO dbus_value_changed: com.victronenergy.system /Dc/Battery/Soc 56
2023-06-03 11:45:55,655 root INFO dbus_value_changed: com.victronenergy.system /Dc/Battery/Soc 57
2023-06-03 11:52:49,714 root INFO Start
2023-06-03 11:52:49,716 root INFO Connected to dbus, and switching over to gobject.MainLoop() (= event based)
2023-06-03 11:52:49,797 dbusmonitor INFO ===== Search on dbus for services that we will monitor starting... =====
2023-06-03 11:52:49,802 dbusmonitor INFO Found: com.victronenergy.hub4, scanning and storing items
2023-06-03 11:52:49,807 dbusmonitor INFO        com.victronenergy.hub4 has device instance 0
2023-06-03 11:52:49,815 dbusmonitor INFO Found: com.victronenergy.settings, scanning and storing items
2023-06-03 11:52:49,818 dbusmonitor INFO        com.victronenergy.settings has device instance 0
2023-06-03 11:52:49,886 dbusmonitor INFO Found: com.victronenergy.system, scanning and storing items
2023-06-03 11:52:49,892 dbusmonitor INFO        com.victronenergy.system has device instance 0
2023-06-03 11:52:49,935 dbusmonitor INFO Found: com.victronenergy.acload.http_47, scanning and storing items
2023-06-03 11:52:49,941 dbusmonitor INFO        com.victronenergy.acload.http_47 has device instance 47
[... cut ...]
2023-06-03 11:52:50,472 root INFO registered ourselves on D-Bus as com.victronenergy.vebus.DTU_id51
2023-06-03 11:52:53,129 root INFO dbus_value_changed: com.victronenergy.hub4 /PvPowerLimiterActive 1
2023-06-03 11:53:23,108 root INFO Inverter 51 on
2023-06-03 11:55:26,713 root INFO dbus_value_changed: com.victronenergy.system /Dc/Battery/Soc 59
2023-06-03 11:59:26,931 root INFO dbus_value_changed: com.victronenergy.system /Dc/Battery/Soc 60
2023-06-03 12:02:52,120 root INFO dbus_value_changed: com.victronenergy.system /Dc/Battery/Soc 61

Telekatz, 2023-06-03:
Da ist ein Fehler beim Verbinden mit dem MQTT Broker. Verwendest du den MQTT Broker von Venus OS oder einen externen Broker?

TomBo1969, 2023-06-03:
Es ist ein externer (Mosquitto auf einem anderen RasPi), ich habe aber den internen im Venus OS aktiviert..

Telekatz, 2023-06-03:
Teste mal den Branch [MQTT_reconnect](https://github.com/Telekatz/venus.dbus-hoymiles/tree/MQTT_reconnect). Nach dem Reboot ein paar Minuten warten und dann sehen, ob sich wieder alles verbindet.

TomBo1969, 2023-06-04:
Das hat leider nicht funktioniert (siehe current.log)
Ich hatte sogar erstmalig den Effekt, dass ich gar nicht mehr auf das Venus OS zugreifen konnte  nach einem Soft-Reboot (Remote-Console, VRM oder SSH). Erst nach erneutem stromlos machen kam ich wieder aufs System.

Auszug aus current.log (Systemzeit hat sich durch Netzwerkproblem auch verschluckt):

2023-06-04 09:44:00,481 root INFO dbus_value_changed: com.victronenergy.hub4 /PvPowerLimiterActive 1
2023-06-04 09:44:05,476 root INFO dbus_value_changed: com.victronenergy.hub4 /PvPowerLimiterActive 0
2023-06-04 09:44:31,896 dbusmonitor INFO Found: com.victronenergy.acload.http_47, scanning and storing items
2023-06-04 09:44:31,912 dbusmonitor INFO        com.victronenergy.acload.http_47 has device instance 47
2023-06-04 09:44:31,967 root INFO dbus device added: com.victronenergy.acload.http_47 47
2023-06-04 09:44:36,10 root INFO dbus_value_changed: com.victronenergy.hub4 /PvPowerLimiterActive 1
2023-06-04 09:42:55,317 root INFO Start
2023-06-04 09:42:55,323 root INFO Connected to dbus, and switching over to gobject.MainLoop() (= event based)
2023-06-04 09:42:55,373 dbusmonitor INFO ===== Search on dbus for services that we will monitor starting... =====
2023-06-04 09:42:55,377 dbusmonitor INFO Found: com.victronenergy.settings, scanning and storing items
2023-06-04 09:42:55,382 dbusmonitor INFO        com.victronenergy.settings has device instance 0
2023-06-04 09:42:55,468 dbusmonitor INFO Found: com.victronenergy.system, scanning and storing items
2023-06-04 09:42:55,473 dbusmonitor INFO        com.victronenergy.system has device instance 0
2023-06-04 09:42:55,517 dbusmonitor INFO ===== Search on dbus for services that we will monitor finished =====
2023-06-04 09:42:55,677 root INFO Setting /Settings/SystemSetup/AcInput1 does not exist yet or must be adjusted
2023-06-04 09:42:55,753 root INFO Setting /Settings/SystemSetup/AcInput2 does not exist yet or must be adjusted
2023-06-04 09:42:55,836 root INFO registered ourselves on D-Bus as com.victronenergy.hm
2023-06-04 09:42:55,989 root ERROR Fehler beim connecten mit Broker
Traceback (most recent call last):
  File "/data/dbus-hoymiles/HMpvinverter.py", line 581, in _init_MQTT
    self._MQTTclient.connect(self.settings['/MqttUrl'])  # connect to broker
  File "/usr/lib/python3.8/site-packages/paho/mqtt/client.py", line 941, in connect
    return self.reconnect()
  File "/usr/lib/python3.8/site-packages/paho/mqtt/client.py", line 1075, in reconnect
    sock = self._create_socket_connection()
  File "/usr/lib/python3.8/site-packages/paho/mqtt/client.py", line 3546, in _create_socket_connection
    return socket.create_connection(addr, source_address=source, timeout=self._keepalive)
  File "/usr/lib/python3.8/socket.py", line 808, in create_connection
    raise err
  File "/usr/lib/python3.8/socket.py", line 796, in create_connection
    sock.connect(sa)
OSError: [Errno 101] Network is unreachable
2023-06-04 09:42:56,56 root INFO registered ourselves on D-Bus as com.victronenergy.vebus.DTU_id51
2023-06-04 09:42:56,148 root INFO setting changed, setting: /Settings/SystemSetup/AcInput1, old: 1, new: 1
2023-06-04 09:42:56,150 root INFO setting changed, setting: /Settings/SystemSetup/AcInput2, old: 0, new: 0
2023-06-04 09:42:56,440 dbusmonitor INFO Found: com.victronenergy.hub4, scanning and storing items
2023-06-04 09:42:56,470 dbusmonitor INFO        com.victronenergy.hub4 has device instance 0
2023-06-04 09:42:56,547 root INFO dbus device added: com.victronenergy.hub4 0
2023-06-04 09:43:21,392 root INFO dbus_value_changed: com.victronenergy.hub4 /PvPowerLimiterActive 1
2023-06-04 09:43:26,398 root INFO dbus_value_changed: com.victronenergy.hub4 /PvPowerLimiterActive 0
2023-06-04 09:43:56,216 root ERROR Fehler beim connecten mit Broker
Traceback (most recent call last):
  File "/data/dbus-hoymiles/HMpvinverter.py", line 486, in _inverterLoop
    self._MQTTclient.connect(self.settings['/MqttUrl'])
  File "/usr/lib/python3.8/site-packages/paho/mqtt/client.py", line 941, in connect
    return self.reconnect()
  File "/usr/lib/python3.8/site-packages/paho/mqtt/client.py", line 1075, in reconnect
    sock = self._create_socket_connection()
  File "/usr/lib/python3.8/site-packages/paho/mqtt/client.py", line 3546, in _create_socket_connection
    return socket.create_connection(addr, source_address=source, timeout=self._keepalive)
  File "/usr/lib/python3.8/socket.py", line 808, in create_connection
    raise err
  File "/usr/lib/python3.8/socket.py", line 796, in create_connection
    sock.connect(sa)
OSError: [Errno 101] Network is unreachable
2023-06-04 09:51:27,352 dbusmonitor INFO Found: com.victronenergy.acload.http_47, scanning and storing items
2023-06-04 09:51:27,428 dbusmonitor INFO        com.victronenergy.acload.http_47 has device instance 47
2023-06-04 09:51:27,463 root INFO dbus device added: com.victronenergy.acload.http_47 47
2023-06-04 09:51:30,264 root INFO dbus_value_changed: com.victronenergy.hub4 /PvPowerLimiterActive 1

Telekatz, 2023-06-04:
Funktioniert es, wenn du den MQTT Broker von Venos OS für den HM-800 verwendest?

TomBo1969, 2023-06-04:
Habe ich bisher nicht getestet, da ich die Daten auch im ioBroker auslese und der MQTT-Adapter dort meiner Kenntnis nach User und Passwörter als Client benötigt. Soll ich mit dem Branch testen oder mit main?
Verwendest du den internen MQTT Broker vom Venus OS?

Telekatz, 2023-06-04:
Teste mit dem main Branch. Ich verwende den MQTT von Venus OS. Der ist so eingestellt, dass kein User oder Passwort benötigt wird.

Telekatz, 2023-06-04:
Hab nochmal im Branch [MQTT_reconnect](https://github.com/Telekatz/venus.dbus-hoymiles/tree/MQTT_reconnect) die MQTT Verbindung überarbeitet. Deinen Fehler "OSError: [Errno 101] Network is unreachable" konnte ich reproduzieren, indem ich das Netzwerkkabel beim starten aus gesteckt hatte. Nach dem wieder einstecken lief mit diese Version das Script dann normal los.

TomBo1969, 2023-06-05:
Jetzt hat es funktioniert. Top, danke!

OSError: [Errno 101] Network is unreachable
2023-06-05 08:03:03,225 dbusmonitor INFO Found: com.victronenergy.acload.http_47, scanning and storing items
2023-06-05 08:03:03,291 dbusmonitor INFO        com.victronenergy.acload.http_47 has device instance 47
2023-06-05 08:03:03,361 root INFO dbus device added: com.victronenergy.acload.http_47 47
2023-06-05 08:03:50,962 root INFO MQTT not connected, try reconnect (ID:51)
2023-06-05 08:03:50,976 root INFO MQTT_connect to 192.168.178.188:1883 rc 0
2023-06-05 08:03:50,984 root INFO MQTT connected (ID:51)
2023-06-05 08:04:20,982 root INFO Inverter 51 on

Verständnisfrage: Wird das demnächst in den main Branch übernommen?

Danke und Gruß, 
Tom

Telekatz, 2023-06-05:
Ja, wird in den main Branch übernommen.

closed 2023-06-05 by TomBo1969