Telekatz/venus.dbus-hoymiles #10: Akku auschalten, wenn voll oder kein Überschuss vorhanden ist

Head, 2023-06-18:
Hi,

hast du schonmal drüber nachgedacht, ob es möglich ist den Akku vollkommen aus zu schalten?
- Wenn er voll ist und der PV-Überschuss eingespeist wird.
- Am Morgen, wenn noch kein Überschuss vorhanden ist und die PV Leistung im Haus verbraucht wird.
- Im Winter oder schlechten Tagen, wenn man ihn dann gerne auf 50% stehen lassen will.

Ich weiß (bisher) nicht, ob es über den CAN-Bus möglich wäre, aber die Kontakt-Schnittstelle scheint ja Pin 1 & 2 dafür zu haben:
[image]

Wenn er voll ist, macht er bei mir dieses hier, glaube nicht, dass das auf Dauer so gut für den Akku ist:
[image]

Hängt vielleicht aber auch damit zusammen, dass der Hoymiles (bei mir) sehr unruhig ist, wenn man das Limit über 200W einstellt, da schwankt die Leistung sekündlich um 50-100W rauf und runter, was der Akku dann natürlich liefert und die Steuerung versucht dagegen zu steuern.

[image]

Wobei es dann auch wieder Bereiche gibt, da ist es wieder stabiler:
[image]

Telekatz, 2023-06-18:
Sehe da irgendwie keine Vorteil, das zu tun. 

Überraschenderweise bringt Venus OS nicht mal einen Fehler, wenn der Akku im Betrieb ausgeschaltet wird. Allerdings schaltet er sich sofort wieder automatisch ein, wenn der Laderegler noch läuft. Nur wenn der Laderegler auch keine PV Energie mehr bekommt, lässt sich der Akku ausschalten.

Head, 2023-06-19:
Interessant. Aber wahrscheinlich geht das so nicht. Ein YouTuber mit Multiplus sagte auch, diese Schwankungen bei 100% sind in jedem System. Finde es halt komisch, beim eAuto soll man nur bis 80% laden und bei der PV-Anlage wird der Akku - wenn er voll ist - den ganzen Tag bei 100% hin und her geladen, wenn auch nur mit wenigen Watt.