Telekatz/venus.dbus-hoymiles #12: Keine Anzeige mehr nach Installation, nur noch weißer Bildschirm

gerd-git, 2023-07-12:
Hallo Telekatz,

nach Installation via:

```
root@raspberrypi2:~# wget https://github.com/telekatz/venus.dbus-hoymiles/archive/refs/heads/main.zip
--2023-07-12 11:32:54--  https://github.com/telekatz/venus.dbus-hoymiles/archive/refs/heads/main.zip
Resolving github.com... 140.82.121.3
Connecting to github.com|140.82.121.3|:443... connected.
HTTP request sent, awaiting response... 302 Found
Location: https://codeload.github.com/Telekatz/venus.dbus-hoymiles/zip/refs/heads/main [following]
--2023-07-12 11:32:55--  https://codeload.github.com/Telekatz/venus.dbus-hoymiles/zip/refs/heads/main
Resolving codeload.github.com... 140.82.121.10
Connecting to codeload.github.com|140.82.121.10|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: unspecified [application/zip]
Saving to: ‘main.zip’

main.zip                                                        [ <=>                                                                                                                                     ] 521.02K  3.01MB/s    in 0.2s

2023-07-12 11:32:55 (3.01 MB/s) - ‘main.zip’ saved [533528]

root@raspberrypi2:~# /data/dbus-hoymiles/uninstall.sh
-sh: /data/dbus-hoymiles/uninstall.sh: No such file or directory

root@raspberrypi2:~# unzip main.zip "venus.dbus-hoymiles-main/*" -d /data
Archive:  main.zip
   creating: venus.dbus-hoymiles-main/
  inflating: venus.dbus-hoymiles-main/HMpvinverter.py
  inflating: venus.dbus-hoymiles-main/LICENSE
  inflating: venus.dbus-hoymiles-main/README.md
  inflating: venus.dbus-hoymiles-main/config.ini
   creating: venus.dbus-hoymiles-main/img/
  inflating: venus.dbus-hoymiles-main/img/inverter.png
  inflating: venus.dbus-hoymiles-main/img/main.png
  inflating: venus.dbus-hoymiles-main/img/settings.png
  inflating: venus.dbus-hoymiles-main/img/system_diagram.png
  inflating: venus.dbus-hoymiles-main/install.sh
   creating: venus.dbus-hoymiles-main/qml/
  inflating: venus.dbus-hoymiles-main/qml/MultiHm.qml
  inflating: venus.dbus-hoymiles-main/qml/OverviewGridParallel.diff
  inflating: venus.dbus-hoymiles-main/qml/PageHmSetup.qml
  inflating: venus.dbus-hoymiles-main/qml/PageMain.diff
  inflating: venus.dbus-hoymiles-main/qml/PageVebusHm.qml
  inflating: venus.dbus-hoymiles-main/qml/overview-inverter-Hm.svg
  inflating: venus.dbus-hoymiles-main/qml/overview-inverter-short-Hm.svg
  inflating: venus.dbus-hoymiles-main/restart.sh
   creating: venus.dbus-hoymiles-main/service/
  inflating: venus.dbus-hoymiles-main/service/run
  inflating: venus.dbus-hoymiles-main/uninstall.sh
root@raspberrypi2:~# mv /data/venus.dbus-hoymiles-main /data/dbus-hoymiles
root@raspberrypi2:~# chmod a+x /data/dbus-hoymiles/install.sh
root@raspberrypi2:~# /data/dbus-hoymiles/install.sh
patching file /opt/victronenergy/gui/qml/PageMain.qml
patching file /opt/victronenergy/gui/qml/OverviewGridParallel.qml
`root@raspberrypi2:~#
```

wird nichts mehr angezeigt:

[image]

Auch ein 

`root@raspberrypi2:~# reboot
`
 brachte nix.

Glücklicherweise ist wieder alles ok nach zweimal uninstall.sh, jedoch gab es Fehlermeldungen:

```
root@raspberrypi2:~# /data/dbus-hoymiles/uninstall.sh
/data/dbus-hoymiles/uninstall.sh: line 8: ./restart.sh: No such file or directory
root@raspberrypi2:~# /data/dbus-hoymiles/uninstall.sh
rm: can't remove '/service/dbus-hoymiles': No such file or directory
kill: usage: kill [-s sigspec | -n signum | -sigspec] pid | jobspec ... or kill -l [sigspec]
/data/dbus-hoymiles/uninstall.sh: line 8: ./restart.sh: No such file or directory
cp: can't stat '/opt/victronenergy/gui/qml/OverviewGridParallel._qml': No such file or directory
cp: can't stat '/opt/victronenergy/gui/qml/PageMain._qml': No such file or directory
rm: can't remove '/opt/victronenergy/gui/qml/OverviewGridParallel._qml': No such file or directory
rm: can't remove '/opt/victronenergy/gui/qml/PageMain._qml': No such file or directory
rm: can't remove '/opt/victronenergy/gui/qml/MultiHm.qml': No such file or directory
rm: can't remove '/opt/victronenergy/gui/qml/PageHmSetup.qml': No such file or directory
rm: can't remove '/opt/victronenergy/gui/qml/PageVebusHm.qml': No such file or directory
rm: can't remove '/opt/victronenergy/themes/ccgx/images/overview-inverter-Hm.svg': No such file or directory
rm: can't remove '/opt/victronenergy/themes/ccgx/images/overview-inverter-short-Hm.svg': No such file or directory

```

venus os v3.00

Gruß Gerd

Head, 2023-07-12:
Die im 3er Branch funktioniert auch unter v3.10~9 bei mir.
Auch bei der neuen /gui-beta/ von Venus, sieht alles korrekt aus:
[image]

gerd-git, 2023-07-12:
Sieht sehr gut aus. Danke.
Muss ich die Dateien alle einzeln laden oder gibt es auch ein main.zip? Sorry, ich kenn mich mit git nicht so gut aus.

Dann noch ne Frage bzgl. Betrieb neben einen Multiplus.:
Da der Multiplus extrem laut ist, ist meine Idee, den Multiplus Nachts auszuschalten und für die Grundlast einen HM-300 als inverter direkt an der Batterie einzuschalten. Das die beiden nicht gleichzeitig laufen können, kann ich mir vorstellen, aber eine Tag-/Nachtumschaltung sollten doch machbar sein, oder?

Telekatz, 2023-07-12:
Das Archiv zum V3.00 Branch gibt es hier [https://github.com/telekatz/venus.dbus-hoymiles/archive/refs/heads/Venus_OS_V3.00.zip](https://github.com/telekatz/venus.dbus-hoymiles/archive/refs/heads/Venus_OS_V3.00.zip)

> Das die beiden nicht gleichzeitig laufen können, kann ich mir vorstellen, aber eine Tag-/Nachtumschaltung sollten doch machbar sein, oder?

Nein, ein gleichzeitige Installation mit einem Multiplus ist nicht möglich.

gerd-git, 2023-07-12:
Danke.

Und schade, ich muss dann da ne andere Lösung suchen. Ich kann den ja einfach dran klemmen, aber leder zählt das VRM den Grid-verbrauch nicht mehr, wenn der Multiplus aus ist....

Head, 2023-07-13:
Man kann beim Multiplus die Lüfter tauschen, da gibt es auch Firmen, die das anbieten. Kenne mich da aber selbst nicht aus, habe das nur mal gesehen.
Wer hat den Multiplus denn installiert? Woher bekommt der die Grid Daten?

gerd-git, 2023-07-13:
Ja, Lüfter schon getauscht und weich aufgehanden. Außerdem gibt es sogar schon ein Fan Upgrade Kit von Victron zur Nachrüstung, damit das Rattern der Lüfters bei geringer Umdrehung verschwindet. Der Lüfter ist aber immer noch sehr laut bei Volllast, aber das hautsächlich in nur Tagsüber wenn man wacht ist.
Ne, das Problem ist Nachts, wenn alles andere ruhig ist und man schlafen will. Der Multiplus brummt mit 60dB auch bei gringer Last. Trotz Lärmschutzdübel und Terassenpads ist er deutlich im durch die geschlssene Tür zu hören. Ein Victron Garantieantrag auch schon gestellt, aber abgelehnt.

GridDaten werden via EM24 gemessen.

gerd-git, 2023-07-23:
> Das Archiv zum V3.00 Branch gibt es hier https://github.com/telekatz/venus.dbus-hoymiles/archive/refs/heads/Venus_OS_V3.00.zip
> 

Ich wollte das probieren, aber leider wieder nur weißer Bildschirm:

```
root@raspberrypi2:~# wget github.com/telekatz/venus.dbus-hoymiles/archive/refs/heads/Venus_OS_V3.00.zip
root@raspberrypi2:~# unzip Venus_OS_V3.00.zip "venus.dbus-hoymiles-Venus_OS_V3.00/*" -d /data
root@raspberrypi2:~# mv /data/venus.dbus-hoymiles-Venus_OS_V3.00/ /data/dbus-hoymiles
root@raspberrypi2:~# chmod a+x /data/dbus-hoymiles/install.sh
root@raspberrypi2:~# /data/dbus-hoymiles/install.sh
root@raspberrypi2:~# /reboot 
```
[image]

Interessant aber im VRM:
[image]

Uninstall findet restart nicht:
```
root@raspberrypi2:~# /data/dbus-hoymiles/uninstall.sh
/data/dbus-hoymiles/uninstall.sh: line 8: ./restart.sh: No such file or directory
root@raspberrypi2:~#
```
Nach reboot aber wieder alles beim alten...
[image]

Telekatz, 2023-07-23:
Hast du immer noch den Multiplus in deinem Venus OS?

gerd-git, 2023-07-23:
Ich hab den ausgeschaltet. Dann ein Reboot. Danach sollte er doch weg sein....

Telekatz, 2023-07-23:
Hast du nach dem Reboot kontrolliert, ob er auch wirklich nicht mehr in der Geräteliste war?

gerd-git, 2023-07-23:
Ich hab den jetzt auch vom VE.Bus Adapter getrennt. Ich probiere nochmal....

gerd-git, 2023-07-23:
Jetzt sieht es anders aus:
[image]
HM-600 wird erkannt, habe ihn gerade mal erst konfiguriert.
HM-300 fehlt noch:
[image]

Telekatz, 2023-07-23:
Weitere Inverter müssen erst in der config.ini aktiviert werden.

gerd-git, 2023-07-23:
Nach reboot wird nichts mehr angezeigt.

Dann Änderung von config.ini
```
[DEFAULT]
#Possible Options for Log Level: CRITICAL, ERROR, WARNING, INFO, DEBUG, NOTSET
Logging=INFO

[Inverter1]
Deviceinstance=51

 Uncomment for additional devices
 [Inverter2]
 Deviceinstance=52

# [Inverter3]
# Deviceinstance=54
```
install nochmal aufgerufen (oder kann ich den service irgendwie anders starten?)
jetzt keine Inverter mehr da:
[image]
[image]

Änderung von config.ini rückgängig gemacht:
install.sh aufgerufen, KEIN reboot
jetzt:
[image]
[image]

HM-300 fehlt.

gerd-git, 2023-07-23:
[image]

Telekatz, 2023-07-23:
Probier's mal so:

```
[DEFAULT]
#Possible Options for Log Level: CRITICAL, ERROR, WARNING, INFO, DEBUG, NOTSET
Logging=INFO

[Inverter1]
Deviceinstance=51

#Uncomment for additional devices
[Inverter2]
Deviceinstance=52

# [Inverter3]
# Deviceinstance=54
```

gerd-git, 2023-07-23:
Klar hatte Tomaten auf den Augen...

```
root@raspberrypi2:~# cat /data/dbus-hoymiles/config.ini
[DEFAULT]
#Possible Options for Log Level: CRITICAL, ERROR, WARNING, INFO, DEBUG, NOTSET
Logging=INFO

[Inverter1]
Deviceinstance=51

# Uncomment for additional devices
[Inverter2]
Deviceinstance=52

# [Inverter3]
# Deviceinstance=54
root@raspberrypi2:~# /data/dbus-hoymiles/install.sh
ln: /service/dbus-hoymiles/service: File exists
patching file /opt/victronenergy/gui/qml/PageMain.qml
patching file /opt/victronenergy/gui/qml/OverviewGridParallel.qml
root@raspberrypi2:~#

```

Trotzdem feht der HM-300 immer noch:

[image]

Telekatz, 2023-07-23:
Reboot gemacht?

gerd-git, 2023-07-23:
Danke, das sieht gut aus:
[image]

Jetzt noch Überschusseinspeisung konfigurieren.... Tipp?

Telekatz, 2023-07-23:
Einfach mal die Einstellungen ausprobieren. 

Du solltest aber auch darauf achten, nicht schneller zu regeln, als die Leistung von den Invertern aktualisiert wird. Wenn beide Inverter über die selbe DTU gesteuert werden verdoppelt sich die Zeit, in der die Aktualisierung erfolgt. Für eine schnellere Regelung würde ich ein externes Powermeter für die Inverter empfehlen.

gerd-git, 2023-07-23:
Ich probiere schon rum.

Es gibt drei Varianten: Grid Target, Base Load, Venus OS. 
Problem bei allen Varianten: Es wird auf 50W statt auf 0W geregelt. 
[image]
Setze ich den Sollwert auf -50W passt es:
[image]
[image]

Als wenn einer sich verrechnet....

gerd-git, 2023-07-23:
Der HM-600 ändert sein Limit, der HM-300 bleibt aber auf 5% festgenagelt.
[image]
[image]

Telekatz, 2023-07-23:
Dass der HM-300 nicht regelt ist so beabsichtigt. Es wird  Primär mit dem ersten Inverter die Last ausgeregelt. Die weiteren Inverter werden erst dann nach geregelt, wenn mit dem ersten Inverter die Last nicht eingestellt werden kann.

Hast du in deinem System weitere AC gekoppelte Inverter?

gerd-git, 2023-07-23:
Nein. Keine weiteren AC-Inverter. Und der Multiplus ist immer noch aus.

Telekatz, 2023-07-23:
Du hast ein Problem beim Anschluss deines HM-600 an die Batterie. Die Beiden Kanäle liefern nicht die gleiche Leistung. Die Minus Leitungen vom Inverter zur Batterie müssen exakt gleich lang sein, sonst fließt der Strom nicht symmetrisch in beiden Kanälen und die Regelung passt dann nicht mehr.

gerd-git, 2023-07-23:
Richtig, die einen sind 40cm länger. Ändere ich sofort.....

gerd-git, 2023-07-23:
In der Tat:
[image]
Jetzt passt auch die Nulleinspeisung:
[image]
[image]

Ich bin mir noch nicht sicher, wo die Vor- und Nachteile der drei Varianten sind. 
Kann ich das auf "Venus OS" stehen lassen? Ist ESS immer vorhanden, auch wenn der Multiplus nicht mehr da ist?

gerd-git, 2023-07-23:
Und leider gibt es auch bei Feed-In Mode kein "OFF". Ich würde gerne via NodeRed + MQTT das Limit selber regeln.... Oder übersehe ich da was?

Telekatz, 2023-07-23:
> Kann ich das auf "Venus OS" stehen lassen? Ist ESS immer vorhanden, auch wenn der Multiplus nicht mehr da ist?

Beim Mode `Venus OS` kommt die Regelung von Venus OS und nicht vom Script wie bei `Grid Target`. Beide regeln den Netzbezug auf den `Sollwert Netz` aus. Dass ESS auch mit dem HM verfügbar ist, ist Ziel diese Scripts.

Die externe Regelung der Leistung ist eine noch offene Baustelle [https://github.com/Telekatz/venus.dbus-hoymiles/issues/4](https://github.com/Telekatz/venus.dbus-hoymiles/issues/4)

Head, 2023-07-24:
@gerd-git man könnte aber auch ohne die offene Baustelle im Venus OS mode selbst regeln über den Grid setpoint im Venus Node-Red. Ich hatte hier mal angefangen und mir dann einen memory leak mit einer NodeRed Addier-Lib eingefangen und musste mein ganzes Venus OS neu aufsetzen. Seit dem keine Muße mehr.

gerd-git, 2023-07-24:
Danke Zusammen. Ich hab das Skript jedoch wieder deinstalliert. Die Regelung ist leider zu langsam um den Multiplus komplett, also auch Tagsüber, ersetzen zu können.

Ansich ist der Multiplus ja genau das was man so braucht. Wenn er nicht so Brummen würde. Nachts abschalten via Node-Red ja ist kein Problem. Ebenso das ersatzweise Einschalten meiner Hoymiles, sogar mit einem einfachen für die Nacht ausreichenden Nulleinspeisungsalogritmus (Limit +-5% bei Abweichung von Null alle 15 Sekunden). Wäre für mich Ideal, wenn das VRM bei ausgeschalteten Multiplus weiter den Grid-Verbauch zählen würde, macht es aber nicht. Deswegen versuchte ich das hier. 

Ich bin nun aber zu einen Punkt angekommen, wo mir der Fehler im VRM egal ist. Schließlich ist nur das bedeutend, was auf dem Netzbetreiberzähler am Ende das Jahres steht.

Nochmal Danke für Ihre Unterstützung.