Telekatz/venus.dbus-hoymiles #14: Feed-in PowerLimit will set to 30W, therefore no regulation

MrBig197x, 2023-11-07:
[venus.dbus-hoymiles](https://github.com/Telekatz/venus.dbus-hoymiles)

HM1500 with openDTU is integrated
All settings in the Victron Cerbo are made
Maximum Power Inverter is set to 700W
MQTT Settings are made
Feed in Power Mode is set to VENUS OS
Tasmota as Grid is installed
ESS is konfigured-
The Problem:
With these settings, the Feed-in Power Limit will be set to 30W or 32W?

MrBig197x, 2023-11-07:
With these settings, the Feed-in Power Limit will be set to max 30W or 32W?

MrBig197x, 2023-11-07:
However, I have another problem with the integration of the Tasmota smart mete (henne49/dbus-opendtu)r. The values are not transmitted reliably, the connection drops. Unfortunately, this is not the way to use it.

Telekatz, 2023-11-08:
What are your settings for HM and ESS? Does Venus OS shows the correct power from your grid meter?

MrBig197x, 2023-11-08:
Hey, Tasmota is integrated and the values are okay

[image]

[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

MrBig197x, 2023-11-08:
Hey, Tasmota is integrated and the Values are okay. Intervall 2 seconds.
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

Head, 2023-11-08:
Mach mal `Optimiert (ohne BatteryLife)`, das hatte bei mir auch zu seltsamen Verhalten geführt.

Telekatz, 2023-11-09:
Die Integration deines Grid Meters sieht unvollständig aus. Kein Strom, keine Spannung und nur eine Phase.

MrBig197x, 2023-11-09:
Ich bekomme nur den Total Verbrauch, die aktuelle Wirkleistung saldiert und die aktuelle Wirkleistung L1, L2 und L3 übertragen, Strom und Spannung leider nicht.

Telekatz, 2023-11-09:
In deinem VRM Screenshot steht aber nur die L1 Leistung vom Grid Meter.

Stehen in der log Datei im Scriptverzeichniss irgendwelche Fehlermeldungen?

MrBig197x, 2023-11-09:
Muss ich mir heute Abend anschauen, ich melde mich. Vielen Dank für den bisherigen Support

MrBig197x, 2023-11-09:
Hallo,
die Integration scheint des Tasmota scheint tatsächlich nicht richtig zu sein.
Allerdings kann ich mich nicht mehr per Putty bzw. WINSCP auf den Cerbo aufschalten.
Ich bekomme trotz gesetztem Superuser und Passwort den Zugriff verweigert?
Keine Ahnung warum.
Per Kommandozeile ssh root@xxx.xxx.xxx.xx komme ich drauf.

MrBig197x, 2023-11-10:
Hallo, 
kurze Rückmeldung. 
Das Problem ist gelöst und es war die Tasmota Integration. 
Ich habe den Tasmota über das Projekt venus.dbus-tasmota-smartmeter integriert. 
Vielen Dank für die Hilfe.

closed 2023-11-11 by Telekatz