Telekatz/venus.dbus-hoymiles #15: Akku wird zu leer gesaugt, BMS trennt Akku vom System

DannyGq, 2024-02-06:
Hallo, zunächst möchte ich mich bei dir bedanken für das tolle Projekt. Ich nutze es mit einem div 14kw lifepo4 Akku mit Seplos BMS.
2 Victron Laderegler + Cerbo + Victron Shunt.
Da die Steuerung über das Seplos BMS alles andere als genau ist, habe ich mich entschlossen zusätzlich einen Victron Shunt dafür einzusetzen, was jetzt auch spürbar besser geworden ist. Trotzdem verrechnet sich der Shunt von Zeit zu Zeit, was dazu führt, das die Batterien viel zu leer gezogen wird. Deshalb möchte ich dich bitten, eine Batterie Steuerung hinzu zu fügen, welche dafür sorgt, das der hoymiles Wechselrichter ab einer eingestellten Spannung (Volt) abgeschaltet wird. In der Konfiguration von einem Muliplus ist so eine Einstellung möglich. Mfg

Telekatz, 2024-02-07:
Ich kann das in die nächste Version einbauen. Wird aber noch etwas dauern, da ich gerade größere Änderungen am Code vornehme und das erst noch testen muss.

DannyGq, 2024-02-07:
Ich danke dir schon mal im voraus.

Telekatz, 2024-03-03:
Ich hab die Funktion jetzt in die neue Version im [DEV Branch](https://github.com/Telekatz/venus.dbus-hoymiles/tree/dev) eingebaut.

DannyGq, 2024-03-05:
Habe heute die dev Version installiert, irgendwas stimmt da nicht, egal was ich einstelle der hoymiles startet nicht. Welche cerbo Version ist mindestens erforderlich? Habe die Version 3.13 installiert. Ich packe dir meinen log in den Anhang. Danke

[current.log.txt](https://github.com/Telekatz/venus.dbus-hoymiles/files/14499779/current.log.txt)

Telekatz, 2024-03-06:
Teste bitte mal die neue Version.

DannyGq, 2024-03-24:
Sory das ich mich erst jetzt zurück melde.
Meine Venus ist kaputt gegangen (Kurzschluss) und ich musste diese erst tauschen.
Habe zwischenzeitlich auf einem Raspberry deinen neuen Treiber installiert, und bin begeistert.
Danke dir, genau so hab ich mir das vorgestellt.

DannyGq, 2024-04-05:
Ich mach hier jetzt zu.
Danke.

closed 2024-04-05 by DannyGq