Telekatz/venus.dbus-hoymiles #16: Problem after update to Venus os 3.20

the-rollux, 2024-02-14:
No communication with hoymiles.
Uninstall/reinstall has no effect.

Telekatz, 2024-02-14:
I tested the script with the v3.20 release and it still works.

Are you using the Venus OS MQTT server and have you checked that the DTU connects to it?

the-rollux, 2024-02-15:
current.log

Firmware 3.13
2024-02-15 07:59:55,111 root INFO registered ourselves on D-Bus as com.victronenergy.hm
2024-02-15 07:59:55,338 root INFO MQTT_connect to 127.0.0.1:1883 rc 0
2024-02-15 07:59:55,340 root INFO MQTT connected (ID:52)
2024-02-15 07:59:55,366 root INFO registered ourselves on D-Bus as com.victronenergy.vebus.DTU_id52
2024-02-15 07:59:55,632 root INFO MQTT_connect to 127.0.0.1:1883 rc 0
2024-02-15 07:59:55,646 root INFO MQTT connected (ID:51)
2024-02-15 07:59:55,654 root INFO registered ourselves on D-Bus as com.victronenergy.vebus.DTU_id51

Firmware 3.20
2024-02-14 11:03:12,185 root INFO registered ourselves on D-Bus as com.victronenergy.hm
2024-02-14 11:03:12,358 root INFO MQTT_connect to 127.0.0.1:1883 rc 0
2024-02-14 11:03:12,360 root INFO Client Got Disconnected rc 1
2024-02-14 11:03:12,362 root INFO Unexpected MQTT disconnection. Will auto-reconnect
2024-02-14 11:03:12,362 root INFO Trying to Reconnect
2024-02-14 11:03:12,379 root INFO registered ourselves on D-Bus as com.victronenergy.vebus.DTU_id51
2024-02-14 11:03:13,370 root INFO Client Got Disconnected rc 1
2024-02-14 11:03:13,371 root INFO Unexpected MQTT disconnection. Will auto-reconnect
2024-02-14 11:03:13,372 root INFO Trying to Reconnect

MQTT Explorer can connect to venus mqtt .

==>
https://community.victronenergy.com/questions/262226/venus-os-320-and-mqtt-fails.html￼

Head, 2024-02-15:
> I tested the script with the v3.20 release and it still works.
> 
> Are you using the Venus OS MQTT server and have you checked that the DTU connects to it?

That's also an issue I run into with v3.20~47.
My Shellies and ESP8266 Grid "Lesekopf" can connect to the Venus OS MQTT, but openDTU and Uptime Kuma can't. Kuma can TCP to Port 1883, but not MQTT.
There are some updates since ~47 though, that I haven't tested.
Also it's not related to this script here.

Telekatz, 2024-02-16:
I think I found the problem. The new MQTT server on Venus OS v3.20 does not accept empty strings as user/password. You can either try to set any user/password for the MQTT server in the script or use the new release of the script.

the-rollux, 2024-02-20:
Moin

Damit sollte das letzte Problem erledigt sein ;-)

Change log:

v3.21 – Feb 19th 2024

Changes:

Fix MQTT issue that prevented writing negative numbers over the MQTT protocol, for example to ESS set points. Affects systems with customer-own software. Bug was introduced per v3.20.

closed 2024-03-02 by Telekatz