Telekatz/venus.dbus-hoymiles #18: Second Inverter is always disabled

MrWh1t3-h-brs, 2024-02-28:
I have enabled a second hoymiles inverter in the config.ini (both inverters connected through one OpenDTU). I can see him inside VenusOS and can change his settings. I can see his values too. But unfortunatelly the second inverter receives always the command to shut down from Venus OS although he is enabled in my settings. Even if I restart him manually, a few minutes later he receives a command from Venus OS to shut down. Is that a bug or have I done something wrong?
Thx

Telekatz, 2024-02-28:
Is the startup limit function activated?

MrWh1t3-h-brs, 2024-02-29:
Thanks so far for providing your solution! It fits very well to my needs.

No,
it is deactivated. I provide some pictures maybe it will help.

# Here the both inverters
## Inside Venus OS:
[image]

## config.ini:
[image]

## In OpenDTU:
[image]

# HM-800_2 Setup inside Venus OS:
**Enabled:**       true
**Phase:** L1
**MQTT:** (Not importent here, it works)
**DTU:** OpenDTU
**Startup Limit:** false
**Feed-In Limit Mode:** Grid Target
**Grid Target Interval:** 3s
**Grid Target Power:** 50W
**Grid Target Tolerance Minimum:** 30W
**Grid Target Tolerance Maximum:** 40W
**Power Meter:** Internal
**Restart inverter at midnight:** true

# HM-800 Setup inside Venus OS:
**MQTT Settings:** (not importent here)
**DTU:** OpenDTU

Telekatz, 2024-02-29:
Can you provide a log with Logging=DEBUG where I can see the shut down of the second inverter.

MrWh1t3-h-brs, 2024-03-01:
I could now solve it for me localy by adding the command

closed 2024-03-02 by Telekatz