Telekatz/venus.dbus-hoymiles #19: svc: warning: unable to chdir to /service/gui: file does not exist

tlamoureux24, 2024-06-01:
Hello,

After running install.sh i got this  :
root@einstein:/data# /data/dbus-hoymiles/install.sh
patching file /opt/victronenergy/gui/qml/PageAcInSetup.qml
svc: warning: unable to chdir to /service/gui: file does not exist
patching file /opt/victronenergy/gui/qml/PageMain.qml
patching file /opt/victronenergy/gui/qml/OverviewGridParallel.qml
svc: warning: unable to chdir to /service/gui: file does not exist

I'm running cerbo gx v3.31

Thanks

Telekatz, 2024-06-01:
Make a restart of the cerbo gx.

tlamoureux24, 2024-06-01:
Thanks for your reply but unfortunately already try reboot many times, reboot the inverter too.
Also try uninstall, reboot, re install and same error.
Also try delete folder, reboot, download, unpack and re install same.
Is the /service/gui supposed to be in the download file ? if so it's not in the packed file.

Thanks.

Telekatz, 2024-06-01:
This error comes from the command that restarts the GUI after changing the GUI. If the restart fails, you can manually restart the GUI on the console with `svc -t /service/gui` or simply restart the cerbo. The installation should still be OK.

tlamoureux24, 2024-06-01:
Ok got it install now, thanks a lot.
But i think i have another problem.

2024-06-01 15:03:27,913 root INFO Connected to dbus, and switching over to gobject.MainLoop() (= event based)
2024-06-01 15:03:29,687 dbusmonitor INFO ===== Search on dbus for services that we will monitor starting... =====
2024-06-01 15:03:29,689 dbusmonitor INFO ===== Search on dbus for services that we will monitor starting... =====
2024-06-01 15:03:29,696 dbusmonitor INFO Found: com.victronenergy.hub4, scanning and storing items
2024-06-01 15:03:29,701 dbusmonitor INFO Found: com.victronenergy.settings, scanning and storing items
2024-06-01 15:03:29,710 dbusmonitor INFO        com.victronenergy.hub4 has device instance 0
2024-06-01 15:03:29,713 dbusmonitor INFO        com.victronenergy.settings has device instance 0
2024-06-01 15:03:29,732 dbusmonitor INFO Found: com.victronenergy.settings, scanning and storing items
2024-06-01 15:03:29,737 dbusmonitor INFO        com.victronenergy.settings has device instance 0
2024-06-01 15:03:29,739 dbusmonitor INFO ===== Search on dbus for services that we will monitor starting... =====
2024-06-01 15:03:29,747 dbusmonitor INFO Found: com.victronenergy.settings, scanning and storing items
2024-06-01 15:03:29,751 dbusmonitor INFO        com.victronenergy.settings has device instance 0
2024-06-01 15:03:29,915 dbusmonitor INFO Found: com.victronenergy.system, scanning and storing items
2024-06-01 15:03:29,924 dbusmonitor INFO        com.victronenergy.system has device instance 0
2024-06-01 15:03:29,933 dbusmonitor INFO Found: com.victronenergy.system, scanning and storing items
2024-06-01 15:03:29,947 dbusmonitor INFO Found: com.victronenergy.system, scanning and storing items
2024-06-01 15:03:29,960 dbusmonitor INFO        com.victronenergy.system has device instance 0
2024-06-01 15:03:29,964 dbusmonitor INFO        com.victronenergy.system has device instance 0
2024-06-01 15:03:30,22 dbusmonitor INFO ===== Search on dbus for services that we will monitor finished =====
2024-06-01 15:03:30,40 dbusmonitor INFO ===== Search on dbus for services that we will monitor finished =====
2024-06-01 15:03:30,52 dbusmonitor INFO ===== Search on dbus for services that we will monitor finished =====
2024-06-01 15:03:30,678 root INFO MQTT_connect to 192.168.1.15:1883 rc 0
2024-06-01 15:03:30,705 root INFO MQTT_connect to 192.168.1.15:1883 rc 0
2024-06-01 15:03:30,717 root INFO MQTT connected (SN:1)
2024-06-01 15:03:30,721 root INFO MQTT connected (SN:2)
2024-06-01 15:03:30,766 root INFO registered ourselves on D-Bus as com.victronenergy.pvinverter.DTU_id41
2024-06-01 15:03:30,738 root ERROR Error at _on_MQTT_message
Traceback (most recent call last):
  File "/data/dbus-hoymiles/Inverter.py", line 648, in _on_MQTT_message
    if k in {'ch0/P_AC','0/power'} and self._dbusservice['/State'] >= 1 and self._role == 'acload':
AttributeError: 'HmInverter' object has no attribute '_dbusservice'
2024-06-01 15:03:30,796 root INFO registered ourselves on D-Bus as com.victronenergy.pvinverter.DTU_id40
2024-06-01 15:03:30,784 root ERROR Error at _on_MQTT_message
Traceback (most recent call last):
  File "/data/dbus-hoymiles/Inverter.py", line 648, in _on_MQTT_message
    if k in {'ch0/P_AC','0/power'} and self._dbusservice['/State'] >= 1 and self._role == 'acload':
AttributeError: 'HmInverter' object has no attribute '_dbusservice'

If you need log in debug let me know.

Thanks.

Telekatz, 2024-06-01:
Please try the new version.

tlamoureux24, 2024-06-01:
Uninstall the old using uninstall.sh (with the same error about service/gui
deleting all files and folders
Reboot cerbo gx
Install the new version (still same error with service/gui)

Log output (not the same error)
2024-06-01 16:14:29,735 root INFO Connected to dbus, and switching over to gobject.MainLoop() (= event based)
2024-06-01 16:14:30,892 dbusmonitor INFO ===== Search on dbus for services that we will monitor starting... =====
2024-06-01 16:14:30,903 dbusmonitor INFO Found: com.victronenergy.settings, scanning and storing items
2024-06-01 16:14:30,915 dbusmonitor INFO        com.victronenergy.settings has device instance 0
2024-06-01 16:14:30,930 dbusmonitor INFO ===== Search on dbus for services that we will monitor starting... =====
2024-06-01 16:14:30,938 dbusmonitor INFO Found: com.victronenergy.settings, scanning and storing items
2024-06-01 16:14:30,942 dbusmonitor INFO ===== Search on dbus for services that we will monitor starting... =====
2024-06-01 16:14:30,943 dbusmonitor INFO        com.victronenergy.settings has device instance 0
2024-06-01 16:14:30,950 dbusmonitor INFO Found: com.victronenergy.hub4, scanning and storing items
2024-06-01 16:14:30,956 dbusmonitor INFO        com.victronenergy.hub4 has device instance 0
2024-06-01 16:14:30,967 dbusmonitor INFO Found: com.victronenergy.settings, scanning and storing items
2024-06-01 16:14:30,971 dbusmonitor INFO        com.victronenergy.settings has device instance 0
2024-06-01 16:14:31,95 dbusmonitor INFO Found: com.victronenergy.system, scanning and storing items
2024-06-01 16:14:31,118 dbusmonitor INFO        com.victronenergy.system has device instance 0
2024-06-01 16:14:31,123 dbusmonitor INFO Found: com.victronenergy.system, scanning and storing items
2024-06-01 16:14:31,139 dbusmonitor INFO Found: com.victronenergy.system, scanning and storing items
2024-06-01 16:14:31,149 dbusmonitor INFO        com.victronenergy.system has device instance 0
2024-06-01 16:14:31,155 dbusmonitor INFO        com.victronenergy.system has device instance 0
2024-06-01 16:14:31,210 dbusmonitor INFO ===== Search on dbus for services that we will monitor finished =====
2024-06-01 16:14:31,229 dbusmonitor INFO ===== Search on dbus for services that we will monitor finished =====
2024-06-01 16:14:31,270 dbusmonitor INFO ===== Search on dbus for services that we will monitor finished =====
2024-06-01 16:14:31,791 root INFO MQTT_connect to 192.168.1.15:1883 rc 0
2024-06-01 16:14:31,799 root INFO MQTT connected (SN:1)
2024-06-01 16:14:31,878 root INFO registered ourselves on D-Bus as com.victronenergy.pvinverter.DTU_id40
2024-06-01 16:14:31,955 root INFO MQTT_connect to 192.168.1.15:1883 rc 0
2024-06-01 16:14:31,980 root INFO MQTT connected (SN:2)
2024-06-01 16:14:31,984 root INFO registered ourselves on D-Bus as com.victronenergy.pvinverter.DTU_id41
2024-06-01 16:14:32,68 root ERROR Error at _on_MQTT_message
Traceback (most recent call last):
  File "/data/dbus-hoymiles/Inverter.py", line 652, in _on_MQTT_message
    if k in {'ch0/P_AC','0/power'} and self._dbusservice['/State'] >= 1 and self._role == 'acload':
  File "/opt/victronenergy/dbus-systemcalc-py/ext/velib_python/vedbus.py", line 154, in __getitem__
    return self._dbusobjects[path].local_get_value()
KeyError: '/State'

Really thank you for the support

Telekatz, 2024-06-01:
New version, new try.

tlamoureux24, 2024-06-01:
Still warning when using install.sh but no more error in log now.
I set the limiter and it does nothing (should it change the value in opendtu through mqtt ?) no limit action on grid export

Here the log in debug mode (i replace serial numbers with XXX and YYY)

2024-06-01 17:33:46,506 root INFO Connected to dbus, and switching over to gobject.MainLoop() (= event based)
2024-06-01 17:33:47,729 dbusmonitor INFO ===== Search on dbus for services that we will monitor starting... =====
2024-06-01 17:33:47,736 dbusmonitor INFO Found: com.victronenergy.settings, scanning and storing items
2024-06-01 17:33:47,768 dbusmonitor INFO        com.victronenergy.settings has device instance 0
2024-06-01 17:33:47,776 dbusmonitor INFO ===== Search on dbus for services that we will monitor starting... =====
2024-06-01 17:33:47,784 dbusmonitor INFO Found: com.victronenergy.hub4, scanning and storing items
2024-06-01 17:33:47,789 dbusmonitor INFO ===== Search on dbus for services that we will monitor starting... =====
2024-06-01 17:33:47,793 dbusmonitor INFO        com.victronenergy.hub4 has device instance 0
2024-06-01 17:33:47,800 dbusmonitor INFO Found: com.victronenergy.settings, scanning and storing items
2024-06-01 17:33:47,805 dbusmonitor INFO        com.victronenergy.settings has device instance 0
2024-06-01 17:33:47,809 dbusmonitor INFO Found: com.victronenergy.settings, scanning and storing items
2024-06-01 17:33:47,813 dbusmonitor INFO        com.victronenergy.settings has device instance 0
2024-06-01 17:33:47,960 dbusmonitor INFO Found: com.victronenergy.system, scanning and storing items
2024-06-01 17:33:47,967 dbusmonitor INFO        com.victronenergy.system has device instance 0
2024-06-01 17:33:48,20 dbusmonitor INFO Found: com.victronenergy.system, scanning and storing items
2024-06-01 17:33:48,35 dbusmonitor INFO Found: com.victronenergy.system, scanning and storing items
2024-06-01 17:33:48,47 dbusmonitor INFO        com.victronenergy.system has device instance 0
2024-06-01 17:33:48,50 dbusmonitor INFO        com.victronenergy.system has device instance 0
2024-06-01 17:33:48,56 dbusmonitor INFO ===== Search on dbus for services that we will monitor finished =====
2024-06-01 17:33:48,58 root DEBUG ===== Settings device init starting... =====
2024-06-01 17:33:48,108 root DEBUG Setting /Settings/Devices/mInv_1/ClassAndVrmInstance found
2024-06-01 17:33:48,129 dbusmonitor INFO ===== Search on dbus for services that we will monitor finished =====
2024-06-01 17:33:48,131 root DEBUG ===== Settings device init starting... =====
2024-06-01 17:33:48,140 dbusmonitor INFO ===== Search on dbus for services that we will monitor finished =====
2024-06-01 17:33:48,145 root DEBUG ===== Settings device init starting... =====
2024-06-01 17:33:48,152 root DEBUG Setting /Settings/Devices/mInv_1/CustomName found
2024-06-01 17:33:48,174 root DEBUG Setting /Settings/Devices/mPlus_0/ClassAndVrmInstance found
2024-06-01 17:33:48,184 root DEBUG Setting /Settings/Devices/mInv_1/MaxPower found
2024-06-01 17:33:48,194 root DEBUG Setting /Settings/Devices/mInv_2/ClassAndVrmInstance found
2024-06-01 17:33:48,211 root DEBUG Setting /Settings/Devices/mPlus_0/StartLimit found
2024-06-01 17:33:48,215 root DEBUG Setting /Settings/Devices/mInv_1/Phase found
2024-06-01 17:33:48,223 root DEBUG Setting /Settings/Devices/mInv_2/CustomName found
2024-06-01 17:33:48,243 root DEBUG Setting /Settings/Devices/mPlus_0/StartLimitMin found
2024-06-01 17:33:48,246 root DEBUG Setting /Settings/Devices/mInv_1/MqttUrl found
2024-06-01 17:33:48,257 root DEBUG Setting /Settings/Devices/mInv_2/MaxPower found
2024-06-01 17:33:48,271 root DEBUG Setting /Settings/Devices/mPlus_0/StartLimitMax found
2024-06-01 17:33:48,282 root DEBUG Setting /Settings/Devices/mInv_1/MqttPort found
2024-06-01 17:33:48,291 root DEBUG Setting /Settings/Devices/mInv_2/Phase found
2024-06-01 17:33:48,306 root DEBUG Setting /Settings/Devices/mPlus_0/LimitMode found
2024-06-01 17:33:48,311 root DEBUG Setting /Settings/Devices/mInv_1/MqttUser found
2024-06-01 17:33:48,319 root DEBUG Setting /Settings/Devices/mInv_2/MqttUrl found
2024-06-01 17:33:48,341 root DEBUG Setting /Settings/Devices/mPlus_0/PowerMeterInstance found
2024-06-01 17:33:48,347 root DEBUG Setting /Settings/Devices/mInv_1/MqttPwd found
2024-06-01 17:33:48,349 root DEBUG Setting /Settings/Devices/mInv_2/MqttPort found
2024-06-01 17:33:48,369 root DEBUG Setting /Settings/Devices/mPlus_0/GridTargetFastDeviation found
2024-06-01 17:33:48,380 root DEBUG Setting /Settings/Devices/mInv_1/InverterPath found
2024-06-01 17:33:48,386 root DEBUG Setting /Settings/Devices/mInv_2/MqttUser found
2024-06-01 17:33:48,396 root DEBUG Setting /Settings/Devices/mPlus_0/GridTargetFastInterval found
2024-06-01 17:33:48,415 root DEBUG Setting /Settings/Devices/mInv_1/DTU found
2024-06-01 17:33:48,419 root DEBUG Setting /Settings/Devices/mInv_2/MqttPwd found
2024-06-01 17:33:48,428 root DEBUG Setting /Settings/Devices/mPlus_0/GridTargetSlowInterval found
2024-06-01 17:33:48,443 root DEBUG Setting /Settings/Devices/mInv_1/InverterID found
2024-06-01 17:33:48,450 root DEBUG Setting /Settings/Devices/mInv_2/InverterPath found
2024-06-01 17:33:48,458 root DEBUG Setting /Settings/Devices/mPlus_0/BaseLoadPeriod found
2024-06-01 17:33:48,472 root DEBUG Setting /Settings/Devices/mInv_1/Enabled found
2024-06-01 17:33:48,484 root DEBUG Setting /Settings/Devices/mInv_2/DTU found
[... cut ...]
2024-06-01 17:34:08,5 root DEBUG MQTT message solar/XXXXXXXXXXXX/0/powerdc b'245.0'
2024-06-01 17:34:08,14 root DEBUG MQTT message solar/XXXXXXXXXXXX/0/yieldtotal b'805.302'
2024-06-01 17:34:08,22 root DEBUG MQTT message solar/XXXXXXXXXXXX/0/temperature b'23.6'
2024-06-01 17:34:08,25 root DEBUG MQTT message solar/XXXXXXXXXXXX/0/efficiency b'95.061'
2024-06-01 17:34:09,258 root EXTINFO Limit 2 changed: 176.95561566760443
2024-06-01 17:34:09,258 root EXTINFO Limit 1 changed: 174.1443843323956

Thank you

Telekatz, 2024-06-01:
Do you want to use the Hoymiles as PV inverter or as battery inverter?

tlamoureux24, 2024-06-02:
My two hoymiles are use as pv inverter

Telekatz, 2024-06-02:
The power control in PV inverter mode should now also work.

tlamoureux24, 2024-06-02:
Hello,
I can confirm it works now !!!!!
Very big thank you to fix this issue :)

closed 2024-06-02 by tlamoureux24