Telekatz/venus.dbus-hoymiles #20: Maybe a bug

tlamoureux24, 2024-06-11:
Hello,

I don't know if it's really a bug or if it's something which is wanted.
If i set by another way a limitation on my inverters and if the ac injection is not active on cerbo gx, some minutes after i set manually the limitation cerbo remove the limitation.
Maybe the code check from time to time if limitation is not active and that inverters are at 100% ?

Telekatz, 2024-06-11:
The script regularly checks whether the inverter is in the state it expects from the inverter. If the power is changed outside of the script, the power is changed back to the power specified by the script.

tlamoureux24, 2024-06-12:
Thanks for your reply.
When the injection limiter is activated in Cerbo GX I'm completely ok with it.
But when the injection limiter is disabled in the Cerbo GX why ?

Hifioldtimer, 2024-08-11:
@Telekatz:
I have used your integration to show openDTU Hoymiles data at my Venus OS / VRM with setting PV inverter.

After long search, why zero net power mechanismen at openDTU fails every 5 Minutes with powering full limit watt into grid and coming down to target after 1 minute, i have found that your plugin sends the message:
Received MQTT message on topic: openDTU/123456789/cmd/limit_nonpersistent_absolute

I will be happy, if you can offer a switch for read only data, not to send data to openDTU to set limits.

best regards
Hifioldtimer

Telekatz, 2024-10-06:
@Hifioldtimer 
I have deactivated the automatic refresh of the power limit in the current version for the 'pvinverter' role.

closed 2024-10-06 by Telekatz