Telekatz/venus.dbus-hoymiles #21: Nach Update auf VenusOS 3.32 regelt Skript nicht mehr

TomBo1969, 2024-06-11:
Moin,
nach dem Update auf 3.32 bekomme ich das Skript trotz kompletter Neuinstallation nicht mehr zum Laufen.
Sämtliche Geräte sind verbunden, MQTT steht, keine Fehler im Log, Ahoy-DTU aktualisiert, aber egal ob ein festes Grid-Target oder Steuerung über VenusOS eingestellt ist, es werden fix 31W invertiert (Powerlimit über Ahoy-DTU bei 5%) und es findet keine weitere Regelung statt.
Wo kann ich ansetzen?
Anbei Screenshots zur Verdeutlichung:
[image]
[image]
[image]
[image]

MicroPlus:
[image]
[image]
[image]

Die ESS-Einstellungen:
[image]
[image]

Eine Verständnisfrage, da ich von einer alten Version komme und es noch keinen MicroPlus dort gab: Die Inverterleistung wird nur noch beim MicroPlus angezeigt, nicht auch beim HM-800 ESS? Oder ist das ein Zeichen, dass hier etwas nicht stimmt? Wie gesagt, MQTT-Verbindung steht.

Danke für Unterstützung,
Tom

Telekatz, 2024-06-11:
Der MicroPlus zeigt die Inverterleistung aller Inverter an, die zum ESS gehören. In deinem Fall kommt die Leistung vom extra Leistungsmesser, nicht vom Inverter. Die Leistung, die bei den einzelnen Invertern angezeigt wird, ist aber immer die vom Inverter gemessene Leistung. Da beim HM-800 ESS nichts angezeigt wird, der Shelly 1PM ESS trotzdem Leistung misst funktioniert die MQTT Verbindung anscheinend doch nicht. Bekommst du in der Seite des ESS Inverters irgendwelche Werte angezeigt?

TomBo1969, 2024-06-11:
Ich habe im VenusOS (und der Ahoy-DTU) testweise auf den internen MQTT umgestellt, dein Skript und später auch das VenusOS neu gestartet, keine Veränderung. MQTT-Verbindung wird auf beiden Seiten angezeigt, aber im HM-800 ESS werden keine Werte angezeigt.
[image]
[image]
[image]
[image]

Das Log:
2024-06-11 17:01:06,398 root INFO Connected to dbus, and switching over to gobject.MainLoop() (= event based)
2024-06-11 17:01:07,507 dbusmonitor INFO ===== Search on dbus for services that we will monitor starting... =====
2024-06-11 17:01:07,510 dbusmonitor INFO ===== Search on dbus for services that we will monitor starting... =====
2024-06-11 17:01:07,511 dbusmonitor INFO Found: com.victronenergy.settings, scanning and storing items
2024-06-11 17:01:07,513 dbusmonitor INFO        com.victronenergy.settings has device instance 0
2024-06-11 17:01:07,520 dbusmonitor INFO Found: com.victronenergy.hub4, scanning and storing items
2024-06-11 17:01:07,524 dbusmonitor INFO        com.victronenergy.hub4 has device instance 0
2024-06-11 17:01:07,532 dbusmonitor INFO Found: com.victronenergy.settings, scanning and storing items
2024-06-11 17:01:07,534 dbusmonitor INFO        com.victronenergy.settings has device instance 0
2024-06-11 17:01:07,621 dbusmonitor INFO Found: com.victronenergy.system, scanning and storing items
2024-06-11 17:01:07,625 dbusmonitor INFO        com.victronenergy.system has device instance 0
2024-06-11 17:01:07,640 dbusmonitor INFO Found: com.victronenergy.system, scanning and storing items
2024-06-11 17:01:07,645 dbusmonitor INFO        com.victronenergy.system has device instance 0
2024-06-11 17:01:07,682 dbusmonitor INFO ===== Search on dbus for services that we will monitor finished =====
2024-06-11 17:01:07,695 dbusmonitor INFO Found: com.victronenergy.acload.http_47, scanning and storing items
2024-06-11 17:01:07,701 dbusmonitor INFO        com.victronenergy.acload.http_47 has device instance 47
2024-06-11 17:01:07,775 dbusmonitor INFO Found: com.victronenergy.solarcharger.ttyUSB1, scanning and storing items
2024-06-11 17:01:07,781 dbusmonitor INFO        com.victronenergy.solarcharger.ttyUSB1 has device instance 289
2024-06-11 17:01:07,834 dbusmonitor INFO Found: com.victronenergy.grid.http_30, scanning and storing items
2024-06-11 17:01:07,839 dbusmonitor INFO        com.victronenergy.grid.http_30 has device instance 30
2024-06-11 17:01:07,874 dbusmonitor INFO ===== Search on dbus for services that we will monitor finished =====
2024-06-11 17:01:07,956 root INFO registered ourselves on D-Bus as com.victronenergy.acload.DTU_id40
2024-06-11 17:01:08,34 root INFO MQTT_connect to 192.168.178.100:1883 rc 0
2024-06-11 17:01:08,39 root INFO MQTT connected (SN:1)
2024-06-11 17:01:08,261 dbusmonitor INFO Found: com.victronenergy.acload.DTU_id40, scanning and storing items
2024-06-11 17:01:08,265 dbusmonitor INFO        com.victronenergy.acload.DTU_id40 has device instance 40
2024-06-11 17:01:08,284 root INFO dbus device added: com.victronenergy.acload.DTU_id40 40 
2024-06-11 17:01:08,298 root INFO registered ourselves on D-Bus as com.victronenergy.vebus.MicroPlus_id50
2024-06-11 17:01:08,858 root INFO dbus_value_changed: com.victronenergy.system /Ac/In/0/ServiceName com.victronenergy.grid.http_30

Ahoy-DTU:
[image]
[image]

Unter "MQTT Inverter Path" sollte doch das Topic reichen oder braucht es hier einen dezidierten Pfad?

TomBo1969, 2024-06-11:
> Unter "MQTT Inverter Path" sollte doch das Topic reichen oder braucht es hier einen dezidierten Pfad?

Genau das war es! Ich musste den Pfad "garage/HM-800" angeben und nun läuft es...
Sorry für den Wind, aber vielleicht hilft es ja wenigstens anderen.

Danke für die Mühen und die schnelle Reaktion!!
Ich mache hier dicht.

closed 2024-06-11 by TomBo1969