Telekatz/venus.dbus-hoymiles #23: Nach Neustart nur 30W am Inverter

gereic, 2025-01-13:
Hallo,
Ich habe das Problem, wenn ich die Raspberry neu starte, das "Feed in Power Limit" nur 30W hat.
Deinstalliere ich den Treiber und installiere ihn neu ohne neu zu starten funktioniert es.
Mein setup:
Raspberry Pi mit Venus-OS 3.51
Hoymiles HMS-1600T mit Opendtu als Batterie-Wechselrichter
Lifepo4 mit dbus-serialbattery-plugin
Victron MPPT für das laden der Batterie

Kann es sein, dass es etwas mit der Startreihenfolge zu tun hat ?
Die Batterie wird erst nach dem HMS und Microplus angezeigt.

Telekatz, 2025-01-13:
Hast du nach dem Neustart des Raspberry ein paar Minuten gewartet?

gereic, 2025-01-13:
Ja habe ich.
Gibt es irgendwo einen Status oder ein Logfile, wo ich mehr sehen kann ?
Es wäre interessant zu wissen, was das ESS gerade vorgeben möchte und was der Microplus gerade macht.

Telekatz, 2025-01-13:
Es gibt ein LOG File im Script Verzeichniss. Vorher solltes du in der config.ini den Log Level umstellen, damit das LOG ausführlicher wird:
`Logging=EXTINFO`

gereic, 2025-01-14:
Ich habe das Logging eingeschaltet.
Im Log schreibt er z.B.: 2025-01-14 11:57:54,652 root EXTINFO AcPowerSetpoint: -285.0
[image]
[image]

gereic, 2025-01-14:
Hier noch der Screenshot vom HMS-1600T
[image]

gereic, 2025-01-14:
Wenn ich uninstall und wieder install mache, sieht es so aus:
[image]

gereic, 2025-01-14:
Es fehlt also die Zeile Limit 1 changed:

gereic, 2025-01-14:
Was noch auffällt im Log:
2025-01-14 12:02:39,9 root WARNING USING OUTDATED REGISTRATION METHOD!
2025-01-14 12:02:39,10 root WARNING Please set register=False, then call the register method after adding all mandatory paths. See https://github.com/victronenergy/venus/wiki/dbus-api
2025-01-14 12:02:39,14 root INFO registered ourselves on D-Bus as com.victronenergy.acload.DTU_id40
2025-01-14 12:02:39,96 root INFO MQTT_connect to 127.0.0.1:1883 rc 0
2025-01-14 12:02:39,99 root INFO MQTT connected (SN:1)
2025-01-14 12:02:39,244 dbusmonitor INFO Found: com.victronenergy.acload.DTU_id40, scanning and storing items
2025-01-14 12:02:39,255 dbusmonitor INFO        com.victronenergy.acload.DTU_id40 has device instance 40
2025-01-14 12:02:39,258 root INFO dbus device added: com.victronenergy.acload.DTU_id40 40
2025-01-14 12:02:39,259 root EXTINFO acload: com.victronenergy.acload.DTU_id40 HMS-1600T 40
2025-01-14 12:02:39,266 root WARNING USING OUTDATED REGISTRATION METHOD!
2025-01-14 12:02:39,266 root WARNING Please set register=False, then call the register method after adding all mandatory paths. See https://github.com/victronenergy/venus/wiki/dbus-api
2025-01-14 12:02:39,274 root INFO registered ourselves on D-Bus as com.victronenergy.vebus.MicroPlus_id50

gereic, 2025-01-14:
Hier noch das Log vom Restart, wo es nicht geklappt hat.
2025-01-13 21:46:24,934 root WARNING USING OUTDATED REGISTRATION METHOD!
2025-01-13 21:46:24,935 root WARNING Please set register=False, then call the register method after adding all mandatory paths. See https://github.com/victronenergy/venus/wiki/dbus-api
2025-01-13 21:46:24,939 root INFO registered ourselves on D-Bus as com.victronenergy.acload.DTU_id40
2025-01-13 21:46:25,11 root INFO MQTT_connect to 127.0.0.1:1883 rc 0
2025-01-13 21:46:25,22 root INFO MQTT connected (SN:1)
2025-01-13 21:46:25,137 dbusmonitor INFO Found: com.victronenergy.acload.DTU_id40, scanning and storing items
2025-01-13 21:46:25,148 dbusmonitor INFO        com.victronenergy.acload.DTU_id40 has device instance 40
2025-01-13 21:46:25,151 root INFO dbus device added: com.victronenergy.acload.DTU_id40 40
2025-01-13 21:46:25,152 root EXTINFO acload: com.victronenergy.acload.DTU_id40 HMS-1600T 40
2025-01-13 21:46:25,159 root WARNING USING OUTDATED REGISTRATION METHOD!
2025-01-13 21:46:25,160 root WARNING Please set register=False, then call the register method after adding all mandatory paths. See https://github.com/victronenergy/venus/wiki/dbus-api
2025-01-13 21:46:25,168 root INFO registered ourselves on D-Bus as com.victronenergy.vebus.MicroPlus_id50
2025-01-13 21:46:25,320 dbusmonitor INFO Found: com.victronenergy.hub4, scanning and storing items
2025-01-13 21:46:25,329 dbusmonitor INFO        com.victronenergy.hub4 has device instance 0
2025-01-13 21:46:25,332 root INFO dbus device added: com.victronenergy.hub4 0
2025-01-13 21:46:25,335 root EXTINFO acload: com.victronenergy.acload.DTU_id40 HMS-1600T 40
2025-01-13 21:46:25,451 dbusmonitor INFO Found: com.victronenergy.grid.mqtt_gr001_g1, scanning and storing items
2025-01-13 21:46:25,487 dbusmonitor INFO        com.victronenergy.grid.mqtt_gr001_g1 has device instance 1
2025-01-13 21:46:25,490 root INFO dbus device added: com.victronenergy.grid.mqtt_gr001_g1 1
2025-01-13 21:46:25,491 root EXTINFO acload: com.victronenergy.acload.DTU_id40 HMS-1600T 40
2025-01-13 21:46:26,251 root INFO dbus_value_changed: com.victronenergy.system /Ac/In/0/ServiceName com.victronenergy.grid.mqtt_gr001_g1
2025-01-13 21:46:30,262 root EXTINFO MaxFeedInPower: 200000.0
2025-01-13 21:46:43,221 root EXTINFO dbus_value_changed: com.victronenergy.system /Dc/Battery/Soc 75
2025-01-13 21:46:43,372 root EXTINFO Limit 1 changed: 291
2025-01-13 23:00:04,259 root EXTINFO Inverter 1 restart
2025-01-13 23:00:35,262 root EXTINFO AcPowerSetpoint: 540.0
2025-01-13 23:00:40,261 root EXTINFO AcPowerSetpoint: 402.0
2025-01-13 23:00:45,261 root EXTINFO AcPowerSetpoint: 102.0
2025-01-13 23:00:50,262 root EXTINFO AcPowerSetpoint: 27.0
2025-01-13 23:00:52,91 root EXTINFO Inverter 1 off
2025-01-13 23:00:55,261 root EXTINFO AcPowerSetpoint: 9.0
2025-01-13 23:01:00,261 root EXTINFO AcPowerSetpoint: 3.0
2025-01-14 07:45:28,870 root EXTINFO dbus_value_changed: com.victronenergy.system /Dc/Battery/Soc 76
2025-01-14 08:36:09,443 root EXTINFO dbus_value_changed: com.victronenergy.system /Dc/Battery/Soc 77
2025-01-14 09:05:36,906 root EXTINFO dbus_value_changed: com.victronenergy.system /Dc/Battery/Soc 78
2025-01-14 09:28:46,63 root EXTINFO dbus_value_changed: com.victronenergy.system /Dc/Battery/Soc 79
2025-01-14 09:53:21,322 root EXTINFO dbus_value_changed: com.victronenergy.system /Dc/Battery/Soc 80
2025-01-14 10:16:39,508 root EXTINFO dbus_value_changed: com.victronenergy.system /Dc/Battery/Soc 81
2025-01-14 10:33:21,386 root EXTINFO dbus_value_changed: com.victronenergy.system /Dc/Battery/Soc 82
2025-01-14 10:49:37,135 root EXTINFO dbus_value_changed: com.victronenergy.system /Dc/Battery/Soc 83
2025-01-14 11:05:15,945 root EXTINFO dbus_value_changed: com.victronenergy.settings /Settings/CGwacs/BatteryLife/State 2
2025-01-14 11:05:15,956 root EXTINFO dbus_value_changed: com.victronenergy.system /Dc/Battery/Soc 84
2025-01-14 11:05:17,245 root EXTINFO dbus_value_changed: /Hub4/DisableFeedIn 0
2025-01-14 11:05:17,249 root EXTINFO Limit 1 changed: 40.0
2025-01-14 11:05:17,258 root EXTINFO Inverter start (0) 27.5 V
2025-01-14 11:05:17,260 root EXTINFO AcPowerSetpoint: -789.0
2025-01-14 11:05:17,364 root EXTINFO Inverter 1 on
2025-01-14 11:05:22,245 root EXTINFO AcPowerSetpoint: -1380.0
2025-01-14 11:05:27,247 root EXTINFO AcPowerSetpoint: -1536.0
2025-01-14 11:05:32,244 root EXTINFO AcPowerSetpoint: -1584.0

Telekatz, 2025-01-14:
Kannst du mir noch mit dem MQTT Explorer die Werte des Inverters auslesen, so wie hier z.B.:
[image]

gereic, 2025-01-15:
[image]
[image]
[image]

Telekatz, 2025-01-15:
Das sind die MQTT Daten der DTU. Ich meinte die Daten des Inverters in Venus OS.

Telekatz, 2025-01-15:
Kannst du auch noch den Wert von /VebusService suchen. Müsste irgendwo unter /System stehen.

gereic, 2025-01-15:
Meinst du den ?
[image]

Telekatz, 2025-01-15:
Ja, das war der Wert. Aber ich sehe immernoch nicht, wo das Limit hängen bleibt. Ich werde heute Abend mal eine Version mit mehr Log Ausgaben erstellen.

Telekatz, 2025-01-15:
Kannst du bitte den Branch [V3.42](https://github.com/Telekatz/venus.dbus-hoymiles/tree/V3.42) ausprobieren und schauen was dort im Log steht.

gereic, 2025-01-15:
Hier das Log nach dem Neustart:

2025-01-15 21:16:32,422 dbusmonitor INFO com.victronenergy.solarcharger.ttyUSB0 disappeared from the dbus. Removing it from our lists
2025-01-15 21:16:32,423 root INFO dbus device removed: com.victronenergy.solarcharger.ttyUSB0 288
2025-01-15 21:16:32,424 root EXTINFO acload: com.victronenergy.acload.DTU_id40 HMS-1600T 40
2025-01-15 21:16:32,431 dbusmonitor INFO com.victronenergy.settings disappeared from the dbus. Removing it from our lists
2025-01-15 21:16:32,432 root INFO dbus device removed: com.victronenergy.settings 0
2025-01-15 21:16:32,433 root EXTINFO acload: com.victronenergy.acload.DTU_id40 HMS-1600T 40
2025-01-15 21:16:32,435 dbusmonitor INFO com.victronenergy.settings disappeared from the dbus. Removing it from our lists
2025-01-15 21:16:32,478 root ERROR Fehler beim reconnecten mit Broker
Traceback (most recent call last):
  File "/data/dbus-hoymiles/Inverter.py", line 625, in _on_MQTT_disconnect
    client.connect(self.settings['/MqttUrl'],self.settings['/MqttPort'])
  File "/usr/lib/python3.8/site-packages/paho/mqtt/client.py", line 941, in connect
    return self.reconnect()
  File "/usr/lib/python3.8/site-packages/paho/mqtt/client.py", line 1075, in reconnect
    sock = self._create_socket_connection()
  File "/usr/lib/python3.8/site-packages/paho/mqtt/client.py", line 3546, in _create_socket_connection
    return socket.create_connection(addr, source_address=source, timeout=self._keepalive)
  File "/usr/lib/python3.8/socket.py", line 808, in create_connection
    raise err
  File "/usr/lib/python3.8/socket.py", line 796, in create_connection
    sock.connect(sa)
ConnectionRefusedError: [Errno 111] Connection refused
2025-01-15 21:16:32,537 root CRITICAL Error in Retrying to Connect with Broker
2025-01-15 21:16:32,538 root CRITICAL [Errno 111] Connection refused
2025-01-15 21:16:32,556 dbusmonitor INFO com.victronenergy.system disappeared from the dbus. Removing it from our lists
2025-01-15 21:16:32,572 dbusmonitor INFO com.victronenergy.hub4 disappeared from the dbus. Removing it from our lists
2025-01-15 21:16:32,574 root INFO dbus device removed: com.victronenergy.hub4 0
2025-01-15 21:16:32,576 root EXTINFO acload: com.victronenergy.acload.DTU_id40 HMS-1600T 40
2025-01-15 21:16:32,578 dbusmonitor INFO com.victronenergy.grid.mqtt_gr001_g1 disappeared from the dbus. Removing it from our lists
2025-01-15 21:16:32,580 root INFO dbus device removed: com.victronenergy.grid.mqtt_gr001_g1 1
2025-01-15 21:16:32,582 root EXTINFO acload: com.victronenergy.acload.DTU_id40 HMS-1600T 40
2025-01-15 21:16:32,584 dbusmonitor INFO com.victronenergy.system disappeared from the dbus. Removing it from our lists
2025-01-15 21:16:32,586 root INFO dbus device removed: com.victronenergy.system 0
2025-01-15 21:16:32,588 root EXTINFO acload: com.victronenergy.acload.DTU_id40 HMS-1600T 40
2025-01-15 21:16:33,177 root WARNING Process stoped, restart process 1
2025-01-15 21:16:33,187 root WARNING Process stoped, restart process 0
2025-01-15 21:16:33,200 root CRITICAL Error at main
Traceback (most recent call last):
  File "/data/dbus-hoymiles/HMpvinverter.py", line 85, in _startInverter
    newDevice = HmInverter(serial)
  File "/data/dbus-hoymiles/Inverter.py", line 100, in __init__
    self._dbus = dbusconnection()
  File "/data/dbus-hoymiles/Inverter.py", line 55, in dbusconnection
    return SessionBus() if 'DBUS_SESSION_BUS_ADDRESS' in os.environ else SystemBus()
  File "/data/dbus-hoymiles/Inverter.py", line 46, in __new__
    return dbus.bus.BusConnection.__new__(cls, dbus.bus.BusConnection.TYPE_SYSTEM)
  File "/usr/lib/python3.8/site-packages/dbus/bus.py", line 124, in __new__
    bus = cls._new_for_bus(address_or_type, mainloop=mainloop)
dbus.exceptions.DBusException: org.freedesktop.DBus.Error.FileNotFound: Failed to connect to socket /run/dbus/system_bus_socket: No such file or directory
2025-01-15 21:16:33,214 root CRITICAL Error at main
Traceback (most recent call last):
  File "/data/dbus-hoymiles/HMpvinverter.py", line 96, in _startVebus
    newDevice = MicroPlus()
  File "/data/dbus-hoymiles/MicroPlus.py", line 174, in __init__
    self._dbus = dbusconnection()
  File "/data/dbus-hoymiles/MicroPlus.py", line 49, in dbusconnection
    return SessionBus() if 'DBUS_SESSION_BUS_ADDRESS' in os.environ else SystemBus()
  File "/data/dbus-hoymiles/MicroPlus.py", line 40, in __new__
    return dbus.bus.BusConnection.__new__(cls, dbus.bus.BusConnection.TYPE_SYSTEM)
  File "/usr/lib/python3.8/site-packages/dbus/bus.py", line 124, in __new__
    bus = cls._new_for_bus(address_or_type, mainloop=mainloop)
dbus.exceptions.DBusException: org.freedesktop.DBus.Error.FileNotFound: Failed to connect to socket /run/dbus/system_bus_socket: No such file or directory
2025-01-15 21:17:56,136 root INFO Connected to dbus, and switching over to gobject.MainLoop() (= event based)
2025-01-15 21:17:57,374 dbusmonitor INFO ===== Search on dbus for services that we will monitor starting... =====
2025-01-15 21:17:57,381 dbusmonitor INFO ===== Search on dbus for services that we will monitor starting... =====
2025-01-15 21:17:57,385 dbusmonitor INFO Found: com.victronenergy.settings, scanning and storing items
2025-01-15 21:17:57,387 dbusmonitor INFO Found: com.victronenergy.settings, scanning and storing items
2025-01-15 21:17:57,506 dbusmonitor INFO        com.victronenergy.settings has device instance 0
[... 5 lines cut ...]
[... cut ...]
2025-01-15 21:17:58,270 dbusmonitor INFO Found: com.victronenergy.hub4, scanning and storing items
2025-01-15 21:17:58,278 dbusmonitor INFO        com.victronenergy.hub4 has device instance 0
2025-01-15 21:17:58,280 root INFO dbus device added: com.victronenergy.hub4 0
2025-01-15 21:17:58,282 root EXTINFO acload: com.victronenergy.acload.DTU_id40 HMS-1600T 40
2025-01-15 21:18:03,305 root EXTINFO MaxFeedInPower: 200000.0
2025-01-15 21:18:11,698 root EXTINFO Limit 1 changed: 321
2025-01-15 21:18:12,327 root EXTINFO dbus_value_changed: com.victronenergy.system /Dc/Battery/Soc 70

gereic, 2025-01-15:
Ich denke der Fehler kommt, wenn venus-os stopt.
Das müsste dann der Start sein.
2025-01-15 21:17:57,722 root INFO registered ourselves on D-Bus as com.victronenergy.acload.DTU_id40
2025-01-15 21:17:57,797 root INFO MQTT_connect to 127.0.0.1:1883 rc 0
2025-01-15 21:17:57,800 root INFO MQTT connected (SN:1)
2025-01-15 21:17:58,7 dbusmonitor INFO Found: com.victronenergy.acload.DTU_id40, scanning and storing items
2025-01-15 21:17:58,24 dbusmonitor INFO        com.victronenergy.acload.DTU_id40 has device instance 40
2025-01-15 21:17:58,28 root INFO dbus device added: com.victronenergy.acload.DTU_id40 40
2025-01-15 21:17:58,32 root EXTINFO acload: com.victronenergy.acload.DTU_id40 HMS-1600T 40
2025-01-15 21:17:58,39 root WARNING USING OUTDATED REGISTRATION METHOD!
2025-01-15 21:17:58,40 root WARNING Please set register=False, then call the register method after adding all mandatory paths. See https://github.com/victronenergy/venus/wiki/dbus-api
2025-01-15 21:17:58,46 root INFO registered ourselves on D-Bus as com.victronenergy.vebus.MicroPlus_id50
2025-01-15 21:17:58,270 dbusmonitor INFO Found: com.victronenergy.hub4, scanning and storing items
2025-01-15 21:17:58,278 dbusmonitor INFO        com.victronenergy.hub4 has device instance 0
2025-01-15 21:17:58,280 root INFO dbus device added: com.victronenergy.hub4 0
2025-01-15 21:17:58,282 root EXTINFO acload: com.victronenergy.acload.DTU_id40 HMS-1600T 40
2025-01-15 21:18:03,305 root EXTINFO MaxFeedInPower: 200000.0
2025-01-15 21:18:11,698 root EXTINFO Limit 1 changed: 321
2025-01-15 21:18:12,327 root EXTINFO dbus_value_changed: com.victronenergy.system /Dc/Battery/Soc 70
2025-01-15 21:20:15,220 root EXTINFO dbus_value_changed: com.victronenergy.settings /Settings/CGwacs/BatteryLife/State 10
2025-01-15 21:20:15,804 root EXTINFO dbus_value_changed: /Hub4/DisableFeedIn 0
2025-01-15 21:20:15,809 root EXTINFO Limit 1 changed: 40.0
2025-01-15 21:20:15,817 root EXTINFO Inverter start (0) 26.6 V
2025-01-15 21:20:15,821 root EXTINFO AcPowerSetpoint: -174.0
2025-01-15 21:20:16,4 root EXTINFO Inverter 1 on
2025-01-15 21:20:20,805 root EXTINFO AcPowerSetpoint: -300.0
2025-01-15 21:20:25,805 root EXTINFO AcPowerSetpoint: -327.0
2025-01-15 21:20:30,806 root EXTINFO AcPowerSetpoint: -333.0
2025-01-15 21:20:35,805 root EXTINFO AcPowerSetpoint: -339.0
2025-01-15 21:20:40,804 root EXTINFO AcPowerSetpoint: -345.0
2025-01-15 21:20:58,65 root EXTINFO Inverter 1 on
2025-01-15 21:21:08,305 root EXTINFO AcPowerSetpoint: -354.0
2025-01-15 21:21:15,805 root EXTINFO AcPowerSetpoint: -357.0
2025-01-15 21:21:30,805 root EXTINFO AcPowerSetpoint: -360.0

gereic, 2025-01-15:
Ich habe jetzt probiert einfach im __init__ vom inverter.py einen Delay von 30sek. eingebaut.
Damit mit den Pfaden etwas gewartet wird.
Ich vermute, dass ein Pfad noch nicht existiert und dadurch das _handlechangedvalue nicht aufgerufen wird.
Wenn ich ein unstall und wieder install mache, existieren diese ja schon.
Kann das sein ? 

class HmInverter:

  def __init__(self, serial):

    self.settings = None
    self._inverterLoopCounter = 0
    self._deviceinstance = 40
    self._role = 'acload'
    self._serial = serial
    self._inverterData = {}
    self._limitDeviationCounter = 0
    self.need_reinit = False
    self._dbus = dbusconnection()
    self._restartTimer = None
    self._checkState = False
    self._calibrationValues = None
    self._resendTimeout = 0
    self._dbusservice = None
    **_time.sleep(30) # Sleep for 30 seconds_**
    self.init()

Telekatz, 2025-01-15:
Die Ausgabe ist aber nicht aus dem V3.42 Branch. Du musst die Dateien MicroPlus.py und Inverter.py mit denen aus dem Branch austauschen.

gereic, 2025-01-15:
Sorry,

Hier das log. Stimmt das jetzt ?

2025-01-15 21:50:33,833 root EXTINFO acload: com.victronenergy.acload.DTU_id40 HMS-1600T 40
2025-01-15 21:50:33,754 dbusmonitor INFO com.victronenergy.settings disappeared from the dbus. Removing it from our lists
2025-01-15 21:50:33,835 root CRITICAL Error in Retrying to Connect with Broker
2025-01-15 21:50:33,837 root CRITICAL [Errno 111] Connection refused
2025-01-15 21:50:33,836 dbusmonitor INFO com.victronenergy.system disappeared from the dbus. Removing it from our lists
2025-01-15 21:50:33,842 root INFO dbus device removed: com.victronenergy.system 0
2025-01-15 21:50:33,842 root EXTINFO acload: com.victronenergy.acload.DTU_id40 HMS-1600T 40
2025-01-15 21:50:33,844 dbusmonitor INFO com.victronenergy.hub4 disappeared from the dbus. Removing it from our lists
2025-01-15 21:50:33,844 root INFO dbus device removed: com.victronenergy.hub4 0
2025-01-15 21:50:33,845 root EXTINFO acload: com.victronenergy.acload.DTU_id40 HMS-1600T 40
2025-01-15 21:50:33,852 dbusmonitor INFO com.victronenergy.solarcharger.ttyUSB0 disappeared from the dbus. Removing it from our lists
2025-01-15 21:50:33,853 root INFO dbus device removed: com.victronenergy.solarcharger.ttyUSB0 288
2025-01-15 21:50:33,854 root EXTINFO acload: com.victronenergy.acload.DTU_id40 HMS-1600T 40
2025-01-15 21:50:33,854 dbusmonitor INFO com.victronenergy.grid.mqtt_gr001_g1 disappeared from the dbus. Removing it from our lists
2025-01-15 21:50:33,855 root INFO dbus device removed: com.victronenergy.grid.mqtt_gr001_g1 1
2025-01-15 21:50:33,856 root EXTINFO acload: com.victronenergy.acload.DTU_id40 HMS-1600T 40
2025-01-15 21:50:33,882 dbusmonitor INFO com.victronenergy.system disappeared from the dbus. Removing it from our lists
2025-01-15 21:51:50,139 root INFO Connected to dbus, and switching over to gobject.MainLoop() (= event based)
2025-01-15 21:51:51,802 dbusmonitor INFO ===== Search on dbus for services that we will monitor starting... =====
2025-01-15 21:51:51,807 dbusmonitor INFO Found: com.victronenergy.settings, scanning and storing items
2025-01-15 21:51:51,830 dbusmonitor INFO ===== Search on dbus for services that we will monitor starting... =====
2025-01-15 21:51:51,835 dbusmonitor INFO Found: com.victronenergy.settings, scanning and storing items
2025-01-15 21:51:52,31 dbusmonitor INFO        com.victronenergy.settings has device instance 0
2025-01-15 21:51:52,36 dbusmonitor INFO Found: com.victronenergy.system, scanning and storing items
2025-01-15 21:51:52,87 dbusmonitor INFO        com.victronenergy.system has device instance 0
2025-01-15 21:51:52,91 dbusmonitor INFO ===== Search on dbus for services that we will monitor finished =====
2025-01-15 21:51:52,102 dbusmonitor INFO        com.victronenergy.settings has device instance 0
2025-01-15 21:51:52,107 dbusmonitor INFO Found: com.victronenergy.system, scanning and storing items
2025-01-15 21:51:52,149 dbusmonitor INFO        com.victronenergy.system has device instance 0
2025-01-15 21:51:52,153 dbusmonitor INFO Found: com.victronenergy.solarcharger.ttyUSB0, scanning and storing items
2025-01-15 21:51:52,236 dbusmonitor INFO        com.victronenergy.solarcharger.ttyUSB0 has device instance 288
2025-01-15 21:51:52,241 dbusmonitor INFO ===== Search on dbus for services that we will monitor finished =====
2025-01-15 21:51:52,373 root INFO registered ourselves on D-Bus as com.victronenergy.acload.DTU_id40
2025-01-15 21:51:52,424 root INFO MQTT_connect to 127.0.0.1:1883 rc 0
2025-01-15 21:51:52,452 root INFO MQTT connected (SN:1)
2025-01-15 21:51:52,573 dbusmonitor INFO Found: com.victronenergy.acload.DTU_id40, scanning and storing items
2025-01-15 21:51:52,586 dbusmonitor INFO        com.victronenergy.acload.DTU_id40 has device instance 40
2025-01-15 21:51:52,590 root INFO dbus device added: com.victronenergy.acload.DTU_id40 40
2025-01-15 21:51:52,592 root EXTINFO acload: com.victronenergy.acload.DTU_id40 HMS-1600T 40
2025-01-15 21:51:52,616 root INFO registered ourselves on D-Bus as com.victronenergy.vebus.MicroPlus_id50
2025-01-15 21:51:52,732 dbusmonitor INFO Found: com.victronenergy.hub4, scanning and storing items
2025-01-15 21:51:52,768 dbusmonitor INFO        com.victronenergy.hub4 has device instance 0
2025-01-15 21:51:52,771 root INFO dbus device added: com.victronenergy.hub4 0
2025-01-15 21:51:52,772 root EXTINFO acload: com.victronenergy.acload.DTU_id40 HMS-1600T 40
2025-01-15 21:51:53,22 root INFO dbus_value_changed: com.victronenergy.system /Ac/In/0/ServiceName com.victronenergy.vebus.MicroPlus_id50
2025-01-15 21:51:55,204 dbusmonitor INFO Found: com.victronenergy.grid.mqtt_gr001_g1, scanning and storing items
2025-01-15 21:51:55,257 dbusmonitor INFO        com.victronenergy.grid.mqtt_gr001_g1 has device instance 1
2025-01-15 21:51:55,264 root INFO dbus device added: com.victronenergy.grid.mqtt_gr001_g1 1
2025-01-15 21:51:55,269 root EXTINFO acload: com.victronenergy.acload.DTU_id40 HMS-1600T 40
2025-01-15 21:51:55,272 root EXTINFO dbus_value_changed: /Hub4/DisableFeedIn 0
2025-01-15 21:51:56,4 root INFO dbus_value_changed: com.victronenergy.system /Ac/In/0/ServiceName com.victronenergy.grid.mqtt_gr001_g1
2025-01-15 21:51:57,736 root EXTINFO MaxFeedInPower: 200000.0
2025-01-15 21:51:57,736 root EXTINFO _setLimit: 0, 200000.0
2025-01-15 21:51:57,737 root ERROR Error at _setLimit
Traceback (most recent call last):
  File "/data/dbus-hoymiles/MicroPlus.py", line 973, in _setLimit
[... cut ...]
TypeError: '<' not supported between instances of 'NoneType' and 'float'
2025-01-15 21:52:30,235 root EXTINFO AcPowerSetpoint: -171.0
2025-01-15 21:52:30,236 root EXTINFO _setLimit: 171.0, 200000.0
2025-01-15 21:52:30,236 root ERROR Error at _setLimit
Traceback (most recent call last):
  File "/data/dbus-hoymiles/MicroPlus.py", line 973, in _setLimit
    newLimit = min(newLimit, self._dbusmonitor.get_value('com.victronenergy.hub4','/MaxDischargePower'))
TypeError: '<' not supported between instances of 'NoneType' and 'float'

gereic, 2025-01-15:
Aktuell funktioniert es.
Starte noch ein paar mal neu um zu sehen, ob alles passt.
Ich habe in Node-Red noch einen Restart am Hoymiles realisiert, weil es manchmal so ist, das der Inverter einfach nicht einschaltet.
Hier habe ich einfach einen Timeout von 3min., wenn der Wr dann immer noch 0W liefert sende ich einen Restart.
Wäre eine nützliche Sache.