Telekatz/venus.dbus-hoymiles #24: Nach Update auf Venus OS 3.53 und aktuelle venus.debus-hoymiles Version: Frage Deviceinstance in Config.ini

SolarFrosch, 2025-01-19:
Hallo,

nach Update auf Venus OS 3.53 und aktuelle hoymiles Script Version  ist mir aufgefallen, dass in der config.ini der Setup Parameter "Deviceinstance" entfallen ist.
In der alten Scriptversion hatte ich den Wert Deviceinstance=51 vorgegeben.

Frage: Wo gebe ich jetzt im Setting "Deviceinstance=51" vor?

Aktuell steht unter HM-600 -> Geräte -> VRM-Instanz der Wert 40.
Wie kann ich den alten Wert 51 wieder vorgeben?

Weiter ist mir aufgefallen, dass in "current.log" folgende Warnung auftaucht:
WARNING USING OUTDATED REGISTRATION METHOD!

Viele Grüße!

Mein Setup:
Raspberry Pi mit Venus-OS 3.53
Hoymiles HM-600 mit Opendtu als Batterie-Wechselrichter
Pylontech Lifepo4
2 Victron MPPTs für BatterieLadung
Hichi/Tasmota Infrarot Sensor als Leistungsmessgerät

Telekatz, 2025-01-21:
Die Deviceinstance wird jetzt automatisch von Venus OS vergeben. Wenn du sie ändern möchtest, musst du die Datei /data/conf/settings.xml anpassen. Suche dort nach dieser Zeile:
`<ClassAndVrmInstance type="s" default="acload:40" silent="False">acload:40</ClassAndVrmInstance>`

Die Warmeldung wird mit der nächsten Version behoben.

SolarFrosch, 2025-01-22:
Danke für die Antworten und danke auch für das tolle Projekt.

Viele Grüße

closed 2025-01-22 by SolarFrosch